robotpy/robotpy-wpilib #111: Python 3.3 support

virtuald, 2015-01-02:
@PeterJohnson: Should we support this? Or just do 3.4 all the way. If we choose to support 3.4 only, then we should probably state it in the documentation, and perhaps add a check to setup.py to make sure that they're not installing on an old interpreter.

 3.3 seems reasonable, however -- but perhaps more work than we should do at this point. I'm of the opinion we should *not* support 3.2.

Travis-CI build fails 3.3 at https://travis-ci.org/robotpy/robotpy-wpilib/jobs/45656854, it's because we're using weakref.finalize.

PeterJohnson, 2015-01-02:
I don't want to go away from using weakref.finalize(), but if there's a reasonable way to have a reasonably functional fallback I'd be ok with that.

computer-whisperer, 2015-01-02:
Whatever we decide, it should probably apply across the entirery of robotpy, not just core-robotpy. What does frcsim-robotpy require?

virtuald, 2015-01-02:
Technically, robotpy-frcsim could probably work with python 2.7, as pygazebo doesn't have any special requirements. 

Though, it does have a requirement on asyncio, which was first provided in 3.4. However, trollius provides a backport for 2.7 - 3.4.

virtuald, 2015-01-03:
I'm going to explicitly mark it as being Python 3.4 only.

closed 2015-01-03 by virtuald