robotpy/robotpy-wpilib #113: netconsole implementation

virtuald, 2015-01-04:
I've always had problems with our [old implementation](https://github.com/robotpy/robotpy-crio/tree/2014/utilities/netconsole) not always working on some networks... but maybe that's just a problem with the netconsole protocol in general?

Anyways, since it still exists, we should bring it over here. Where does it belong? Should we install as a script with pyfrc or wpilib, or just distribute with the robotpy installer? Maybe it belongs in a tools directory.

aerobit, 2015-01-06:
a tools directory would be nice, if we're going to ever distribute anything else with it.

I think it should be distributed along with robotpy, because netconsole output as part of a robot.py deploy would be useful.

virtuald, 2015-01-07:
After contemplating this, I think what should be done is that netconsole should be distributed as a separate pip-installable package as a separate project. The other tools that we had in robotpy aren't really necessary to have separate from their upstreams now that we're running in a linux environment.

I'm in a meeting at work, so I'll put something together in a few minutes :-D

virtuald, 2015-01-07:
Ok, pushed it into its own package on pypi, with a separate repo at https://github.com/robotpy/pynetconsole

closed 2015-01-07 by virtuald