robotpy/robotpy-wpilib #122: HALInitialize fails with 'Could not initialize HAL'

virtuald, 2015-01-14:
Reproduced this way:
* Start a program using pyfrc deploy mechanism
* Log in via ssh, and start a program manually (actually, I just started ipython and imported wpilib)

I get the following error message, followed by a HALError that says "Could not initialize HAL". 

```
Killing previously running FRC program...
FRC pid 1705 did not die within 110ms. Aborting
```

If I waited a second and ran it again, by then the other process has died and HALInitialize succeeds. I'm not sure if there's a bad side effect that would happen if we just waited and tried to initialize HAL again. Maybe we shouldn't fix this, but only document it.

PeterJohnson, 2015-01-14:
Looking at HAL.cpp, you can call HALInitialize(1) to kill -9 the old process, or HALInitialize(2) to just warn instead of fail.  It sounds like the timeout should just be a bit longer though.

robobenklein, 2015-01-14:
I also get a HALInitialize error.  
I do not however get a notification "Killing previously running FRC program..."  

To fix this, I just run the code once, let it break, then run it again directly afterwards. It runs exactly as intended.

virtuald, 2015-01-14:
@robobenklein do you get the notificatoin when using pyfrc's deploy function, or only when running a program manually?

robobenklein, 2015-01-14:
I only see it when I run the code manually.

virtuald, 2015-02-19:
Users really don't have a good reason to run code manually, I'm going to close this as wont-fix.

closed 2015-02-19 by virtuald