robotpy/robotpy-wpilib #126: Trouble installing. Username and password mixup?

MaisieG, 2015-01-16:
I'm trying to install robotpy and everything seems to be working up until it asks for the username and password. I enter the username and password for our roborio and then "Log in with your NI-Auth credentials." is printed on the terminal. After about 4-6 seconds it then prints this error message

Permission denied (publickey,keyboard-interactive).
Command ['/usr/bin/ssh', 'whea123@roborio-3881.local', 'md5sum opkg_cache/python3_3.4.2_armv7a-vfp-neon.ipk opkg_cache/install_opkg.sh 2> /dev/null; [ -d "opkg_cache" ]; echo $?'] returned non-zero error status 65280

The part, 'whea123@roborio-3881.local', is what worries me as it seems to be using the password (temporarily whea123) as the username. Any advice is appreciated. Thanks is advance!

virtuald, 2015-01-16:
Interesting. Are you installing 2015.0.4? It no longer asks you for a username/password -- just uses 'admin' and ''. What version did you install?

virtuald, 2015-01-16:
Ha, you're right: https://github.com/robotpy/robotpy-wpilib/blob/2015.0.2/installer/installer.py#L331, However, we no longer ask you for a username/password, so this bug is invalid.

I'm willing to open the username/password question to discussion, but I think teams should probably just leave things be and not pretend that the robot is secure. There are a bunch of ways one can control the robot remotely: various default users, anonymous ftp, the DS protocol isn't secure, neither is networktables.

commit e35e8911 (2015-01-16): If we ever re-enabled this, would want to fix this

- Fixes #126
closed 2015-01-16 by virtuald