robotpy/robotpy-wpilib #128: Need to port the SerialPort object

virtuald, 2015-01-16:
https://github.com/robotpy/allwpilib/blob/master/wpilibj/wpilibJavaDevices/src/main/java/edu/wpi/first/wpilibj/SerialPort.java

PeterJohnson, 2015-01-16:
I guess for compatibility, but is there a reason not to use the native Python libraries for serial ports?

virtuald, 2015-01-16:
That's true enough. If we decide not to do it, we probably should note it somewhere in a list of things that we're explicitly not porting.

auxiliary-character, 2015-01-18:
Either a wrapper around native Python library or a port would be appreciated.

virtuald, 2015-01-19:
@auxiliary-character , have you tried [pyserial](http://pyserial.sourceforge.net/)? You can use the robotpy installer to install pip packages on the robot (see [this page](http://robotpy.readthedocs.org/en/latest/installer.html#download) for instructions on how to use the installer).

auxiliary-character, 2015-01-19:
@virtuald That may work. I'm looking at maybe porting the Java version of the [navx-mxp](https://code.google.com/p/navx-mxp/) code, which depends upon the wpilibj SerialPort. Looking through the hardware documentation, I may want to do I2C or SPI from scratch instead, but it would be convenient to just have a SerialPort class that behaves the same as it does in Java.

virtuald, 2015-01-19:
You're more than welcome to do the port and submit a pull request ;)

virtuald, 2015-01-19:
The mxp thing would be a good addition to https://github.com/robotpy/robotpy-wpilib-utilities, actually.

auxiliary-character, 2015-01-19:
Awesome. The scope's a little beyond what I'm used to, and I [might not get much time to test it](http://www.reddit.com/r/FRC/comments/2scrp3/totes_and_hoes_aka_a_meme_that_was_made_during_a/cnpewyk), but I'll definitely send it your way if/when it happens.

virtuald, 2015-01-19:
Lol, good story at the link.

virtuald, 2015-01-19:
Also, if you can do the translation, we can review it pretty easily. As long as it matches the java code reasonably closely, it should be ok.

auxiliary-character, 2015-01-19:
Awesome.

virtuald, 2015-02-18:
@auxiliary-character did you end up getting that to work?

auxiliary-character, 2015-02-18:
We got a [terrible hackish version](https://github.com/DenfeldRobotics4009/2015_Lopez_Jr/blob/master/imu_simple.py) using regexes that doesn't verify checksums, and doesn't include acclerometer or quaternion data. I think we'll work on a proper version post-season.

virtuald, 2015-11-21:
@auxiliary-character did you guys ever make a proper port of the MXP stuff? Our team just got one. :)

computer-whisperer, 2015-11-22:
I started working on a proper port somewhere, though I dont remember how far I got.

auxiliary-character, 2015-11-23:
Yeah, we never did. I have been procrastinating a lot. Should be pretty easy, though; the [I2C/SPI registers](http://www.pdocs.kauailabs.com/navx-mxp/advanced/register-protocol/) are documented pretty well. I imagine it would be pretty similar to your [adxl345 driver](https://github.com/robotpy/robotpy-wpilib/blob/master/wpilib/wpilib/adxl345_i2c.py) in implementation.

virtuald, 2016-01-16:
BTW, I've pushed my initial complete NavX MXP implementation to https://github.com/robotpy/robotpy-wpilib-utilities/tree/mxp .. it's a lot easier this year, as they moved a bunch into the firmware.

virtuald, 2017-02-13:
Decided I don't care about this, so closing.

PR #263 (2017-08-16): Add the WPILib SerialPort object
I don't have hardware to test this on. @dangyogi can you test this and see if it works for you?

* Fixes #128
commit b93f05ca (2017-08-16): Add the WPILib SerialPort object

- Fixes #128
closed 2017-02-13 by virtuald