robotpy/robotpy-wpilib #136: Commands in command groups don't call end(), interupted(), cancel(), etc. when the parent command is canceled.

auxiliary-character, 2015-02-14:
For instance, say we have a command:

```
from wpilib.commands import Command
class DoTheThing(Command):
    def  __init__(self, robot):
        super().__init__()
        self.robot = robot

    def end(self)
        print("I won't be called")

    def interrupted(self):
        self.end()

    def cancel(self):
        self.end()
        super().cancel()
````

And we set it up in a command group:

```
from wpilib.commands import CommandGroup
class DoAllThings(CommandGroup):
    def __init__(self, robot):
        super().__init__()
        self.robot = robot
        self.addSequential(DoTheThing(self.robot))
```

If we initialize the group and stop it...

```
do_all_things = DoAllThings(whatever_we_called_our_robot_instance)
#Go on with life as a robot, chew up time, etc.
do_all_things.cancel()
```

It won't run our DoTheThing.end() command.

virtuald, 2015-02-14:
Thanks for reporting this. I'll try to look at this tonight.

auxiliary-character, 2015-02-14:
Thanks.
Our workaround for now is just to override cancel() on the CommandGroup and have it cancel() all of our commands.

virtuald, 2015-02-15:
Any chance you can generate a small sample that demonstrates the error? I'm trying to reproduce the bug, but it works for me. See this gist for a robot that has a command/command group, and I can cancel the command.

https://gist.github.com/virtuald/f08d5b9493e60bfd2496

auxiliary-character, 2015-02-15:
Hmm, I think I need to play with this some more. i'm seeing what you're seeing in the simulator, but I'm certain we were having problems with it last night on the robot. It may have been a mistake on our part, but I'll try and coax out and isolate what we were experiencing last night when I next have access to the robot.

virtuald, 2015-04-04:
Closing this due to inability to reproduce.

closed 2015-04-04 by virtuald