robotpy/robotpy-wpilib #137: AnalogTrigger + MXP remap fix

virtuald, 2015-02-14:
https://github.com/robotpy/allwpilib/commit/215002e487050bcbb5dfede04e0e8fd4a7c0411a

virtuald, 2015-02-26:
This appears to be a HAL-only change, so if we bumped the HAL version this would work. I don't have a roboRIO available at the moment to test a new HAL on, so I'm loathe to do this after build season has ended.

computer-whisperer, 2015-02-26:
I have access to a couple of unbagged RoboRIOs. If you were to send me the compiled HAL I could probably run some tests with it.

virtuald, 2015-02-26:
The hal's are available at http://www.tortall.net/~robotpy/hal/ . If you change hal_version at the top of https://github.com/robotpy/robotpy-wpilib/blob/master/hal-roborio/setup.py, then it will download the latest hal when building the setup package. You can use `devtools/build_and_deploy.sh` to deploy the version you have in tree.

computer-whisperer, 2015-02-26:
Is there anything in particular that you would like for me to test for? I just confirmed that it starts and runs my team's code just fine, though I haven't run any real hardware with it yet.

computer-whisperer, 2015-10-17:
This can probably be closed, since I assume the HAL will get a major version bump before we are done with RobotPy 2016

closed 2015-10-17 by virtuald