robotpy/robotpy-wpilib #138: CANTalon can crash the robot on CAN errors

virtuald, 2015-02-15:
Because we actually raise an exception when a CAN error is returned from HAL, if there are any kind of CAN errors that occur, the robot code can crash if you forget to do a try/except block. Looking at the JNI version, it appears all the errors are swallowed. 

I remember people complaining about this type of behavior in CANJaguar in Java, and it was removed as a result. Our robot crashed today because of this, and I'm also inclined to swallow the errors, or giving them to the user via some alternate mechanism (log message? error callback?). 

@PeterJohnson do you have any opinions on this?

PeterJohnson, 2015-02-15:
Concur that we should catch this for the user and log a message instead.

virtuald, 2015-02-15:
@PeterJohnson can you review the fix, and implement a fix for CANJaguar too? I don't know enough about the errors there to know whether they should be kept or not.

commit 990718eb (2015-02-15): Only emit a warning on CANTalon errors

- Fixes #138
closed 2015-02-15 by virtuald