robotpy/robotpy-wpilib #141: If threads crash with unhandled exceptions, the driver station should be notified

virtuald, 2015-02-19:
I suspect there's a python mechanism to do this. The DriverStation error reporting mechanism could be used to do this.

Twinters007, 2017-08-11:
Isn't this handled here?
https://github.com/robotpy/robotpy-wpilib/blob/master/wpilib/wpilib/robotbase.py#L189

virtuald, 2017-08-11:
That only applies to the main thread.

Twinters007, 2017-08-11:
Ah. I see

Twinters007, 2017-08-11:
```
import sys
def my_except_hook(exctype, value, traceback):
    if exctype == KeyboardInterrupt:
        print "Handler code goes here"
    else:
        sys.__excepthook__(exctype, value, traceback)
sys.excepthook = my_except_hook
```

auscompgeek, 2017-08-11:
Probably best to save the old `sys.excepthook` in case something else wants an excepthook as well.