robotpy/robotpy-wpilib #143: HAL STATUSFUNC should distinguish between warnings and errors

virtuald, 2015-02-25:
This bug happens particularly when accessing the PDP. See http://www.chiefdelphi.com/forums/showthread.php?t=135145 for details.

closed 2015-02-25 by virtuald

virtuald, 2015-02-26:
This bug may have happened because of https://github.com/robotpy/allwpilib/commit/7bdd91a058457db7b6d294137a8de26d47163e6d ... I can't see what artf4014 is though.

virtuald, 2015-02-26:
@PeterJohnson can you look at the beta tracker and see what artf4014 is?

PeterJohnson, 2015-02-26:
When there are multiple PCMs, reading the solenoid values from any PCM returns the value from PCM0. This is demonstrated in the attached sample programs which set PCM0 solenoid to Reverse and PCM1 solenoid to Forward. Reading both using GetAll returns 2 (reverse) for both PCM 0 and PCM1. It should return 2 for PCM0 and 1 for PCM1.  	

The PCM ID is not used when getting status in hal/ctre/PCM.cpp, so all status is from PCM 0

commit f5ea4270 (2015-02-25): Don't throw errors when HAL emits a warning

- Fixes #143
PR #144 (2015-02-25): Don't throw errors when HAL emits a warning
- Fixes #143

To test this, you can do the following:

```
git clone https://github.com/robotpy/robotpy-wpilib.git
cd robotpy-wpilib
git checkout hal-warning
```

On a linux/OSX system, or if you have MSYS installed on windows, you can run `devtools/build_and_install.sh`. If you don't have either of those, post a comment and I can write something up that will get you going.