robotpy/robotpy-wpilib #148: Running LiveWindow after calling free() on a resource results in "operation on freed port" errors.

computer-whisperer, 2015-03-13:
There doesn't seem to be a mechanism for removing components from LiveWindow after they have been freed. I'm not quite sure how this should be handled. 

One way would be to add a ```removeComponent()``` method to LiveWindow and have every ```free()``` method call it. Another way (possibly a good idea to do in any case) would be to have LiveWindow catch and log exceptions that it receives from the device objects.

@virtuald, @PeterJohnson what do you think?

virtuald, 2015-03-14:
What's the use case here?

computer-whisperer, 2015-03-14:
If at any time in your program you do this:
```
mymotor = wpilib.Talon(0)
mymotor.free()
```
After this, any attempts to enable LiveWindow will crash the DS thread, since it will try to run ```mymotor.set(0)``` after mymotor has been freed

virtuald, 2015-07-08:
Sorry, sounds like (as usual) you have a good point here. I'm not really planning on working on RobotPy until ~Septemberish, but feel free to make a PR to fix this.

I imagine this probably affects WPILib itself too, yes?

computer-whisperer, 2015-07-08:
Yes, i think it would affect upstream. When i fix it, do you think it would be important to try and push something to Java WPILib? Or should we wait to see what changes they plan on making for the 2016 season?

virtuald, 2015-07-08:
If you feel like pushing it upstream, I imagine that would be a good thing. You can either apply for an account on the WPILib collab.net site and submit a branch via gerrit, or you can do a pull request to our allwpilib repository and I can submit the branch for you.

PR #153 (2015-10-17): LiveWindow work: Fix DS thread crashing from exceptions thrown by LV device hooks.
Added a try/except to catch any exceptions thrown by device's `.startLiveWindowMode()` hook.

Also added a `LiveView.removeComponent()` method, and now call it from the `free()` method of every device that has a `LiveWindow.addDevice()` in it's `__init__()`.

This fixes #148
closed 2015-10-17 by virtuald