robotpy/robotpy-wpilib #149: SPI doesn't work?

aerobit, 2015-05-12:
I'm trying to get an SPI Gyro working, but I can't use the transaction function.
https://github.com/robotpy/robotpy-wpilib/commit/13e140140492567426b5a63f11b1ecefd657e46b
With this patch, SPI doesn't crash the robot, but the data I receive is an array full of 0s, seemingly no matter what I do. Any ideas?

virtuald, 2015-05-13:
Looking at the documentation for the function... it's totally screwed up. Probably should file a bug against WPILib to fix their docs and this API.

* Size appears to only be used for the size of the input buffer
* The data received buffer does not have a defined size. I'm guessing that it's a maximum of 8 bytes. The number of bytes received from the function are in the return value, so we're doing that wrong too. I *think* you want to use `ctypes.create_string_buffer` to allocate the receive buffer instead (to the max size... probably 8), and then only return the number of bytes that are actually read.

@PeterJohnson any thoughts on this?

virtuald, 2015-05-13:
Also, while you're at it, if you want to fix https://github.com/robotpy/robotpy-wpilib/issues/62 feel free :)

PeterJohnson, 2015-05-13:
These Python wrappers are indeed buggy; the I2C functions are similarly broken.  The HAL API documentation looks correct to me; the SPI data received buffer is a defined size because for SPI transactions, size represents both the input and output buffer size (remember that on SPI you receive one byte of data at the same time you send a byte).  The way I originally wrote these functions I was expecting to receive/return arrays of integers rather than byte strings (implementation choice).  The two bugs that need fixing are construction of the ctypes array and resizing of the return array (slicing with [0:rv] should work).

However, I'm not sure why you're just getting 0s coming back after fixing the bug.  My suspicion is that you're not sending a correct request or there's some other hardware issue.  Have you tried identical code in C++ or Java?

virtuald, 2015-12-07:
Ari's fix commited in 976db3a4c9da4dd9a75d308df9abaf595ccc5375

closed 2015-12-07 by virtuald