robotpy/robotpy-wpilib #156: (Unintenional?) Type Punning

jkoritzinsky, 2015-11-16:
Hey guys!  We just noticed an interesting silent bug (doesn't cause issues, but probably isn't right) over at RobotDotNet with the Driver Station synchronization.  We have an issue tracking it at robotdotnet/WPILib#67. We also reported it on Gerrit [here](https://usfirst.collab.net/gerrit/#/c/1008/).

You guys have the exact same bug [here](https://github.com/robotpy/robotpy-wpilib/blob/master/wpilib/wpilib/driverstation.py#L67).

We don't have commit access to the allwpilib repository, but I heard that you guys do.  Can you take a look at this as well?

virtuald, 2015-12-07:
I think the fix for this was pushed into WPILib master, so closing this.

closed 2015-12-07 by virtuald

jkoritzinsky, 2015-12-07:
It was pushed.  Thanks!