robotpy/robotpy-wpilib #157: Add PEP 0484 type hinting?

virtuald, 2015-12-16:
This could be useful to combine with @computer-whisperer's API checkers. Or, pyfrc could automatically run it as part of tests before deploying code -- that would be really cool. 

See http://mypy-lang.org/

CrazyPython, 2016-08-25:
@virtuald 

This needs to be implemented NOW. It really helps my IDE a lot. Seriously.

virtuald, 2016-08-26:
I will accept pull requests. 👍

auscompgeek, 2017-07-20:
I feel like possibly taking this up. I'm just thinking, should we use `*.pyi` stub files, or should we just annotate the source?

Twinters007, 2017-07-20:
If you annotate the source it'll show up if you aren't using an IDE and you can view them easily just by going into the file.

virtuald, 2017-07-20:
My opinion is that it should be done in the source, otherwise it'll never get updated.

You can most likely use the API checker stuff to do a lot of the work for you.

auscompgeek, 2017-09-27:
We should also make devtools/convert_javadoc.html convert the type information to type hints too.

virtuald, 2017-09-27:
That's a good idea.

auscompgeek, 2017-12-12:
Just something I've noticed: full-blown type hints don't seem to look very nice when rendered in the docs. They might look nicer with forward-references, but I haven't had a chance to try that.

nhubbard, 2018-04-02:
@virtuald I am working on this in my fork. The effort is few and far in between, but it will eventually be pull-requested into the master branch.

virtuald, 2018-04-02:
@nhubbard awesome. Probably better to send the PRs in batches instead of the whole thing all at once.

virtuald, 2019-01-02:
I think this is mostly done?

closed 2019-01-02 by virtuald