robotpy/robotpy-wpilib #165: Add 2016 example projects

virtuald, 2016-01-02:
Example projects are in the [eclipse plugins repo](https://usfirst.collab.net/gerrit/gitweb?p=eclipse_plugins.git;a=tree;f=edu.wpi.first.wpilib.plugins.java/resources/templates;hb=HEAD;tflink=projects.wpilib/scm.Eclipse_Plugins).

Looks like there's a few samples that we're missing... Gyro, GyroMecanum, Potentiometer, PotentiometerPID, Ultrasonic, UltrasonicPID.

There are also 3 vision projects, which I think are less important: 2015Vision, AxisCameraSample, IntermediateVision.

Twinters007, 2016-01-02:
Eclipse Plugins Repo link seems to be broken

virtuald, 2016-01-02:
It's possible that you need to be logged in to their site to see it. I mirrored it to github [here](https://github.com/virtuald/wpilib_eclipse_plugins/tree/master/edu.wpi.first.wpilib.plugins.java/resources/templates).

virtuald, 2016-09-14:
Updated this for 2017.

Twinters007, 2016-09-14:
what's changed for 2017

virtuald, 2017-01-07:
The example projects are in their own git repo, closing this.

closed 2017-01-07 by virtuald