robotpy/robotpy-wpilib #175: CANTalon needs to be updated

Arhowk, 2016-01-14:
~~~
import hal
import weakref

from .livewindow import LiveWindow
from .livewindowsendable import LiveWindowSendable
from .motorsafety import MotorSafety
from .resource import Resource
from .timer import Timer

__all__ = ["CANTalon"]

def _freeCANTalon(handle):
    hal.TalonSRX_Destroy(handle)

class CANTalon(LiveWindowSendable, MotorSafety):
    """Talon SRX device as a CAN device

    The TALON SRX is designed to instrument all runtime signals periodically.
    The default periods are chosen to support 16 TALONs with 10ms update rate
    for control (throttle or setpoint).  However these can be overridden with
    :meth:`setStatusFrameRate`.

    Likewise most control signals are sent periodically using the
    fire-and-forget CAN API.

    Signals that are not available in an unsolicited fashion are the Close
    Loop gains.  For teams that have a single profile for their TALON close
    loop they can use either the webpage to configure their TALONs once or
    set the PIDF,Izone,CloseLoopRampRate,etc... once in the robot application.
    These parameters are saved to flash so once they are loaded in the TALON,
    they will persist through power cycles and mode changes.

    For teams that have one or two profiles to switch between, they can use
    the same strategy since there are two slots to choose from and the
    ProfileSlotSelect is periodically sent in the 10 ms control frame.

    For teams that require changing gains frequently, they can use the
    soliciting API to get and set those parameters.  Most likely they will
    only need to set them in a periodic fashion as a function of what motion
    the application is attempting.  If this API is used, be mindful of the CAN
    utilization reported in the driver station.

    Encoder position is measured in encoder edges.  Every edge is counted
    (similar to roboRIO 4X mode).  Analog position is 10 bits, meaning 1024
    ticks per rotation (0V => 3.3V).  Use :meth:`setFeedbackDevice` to select
    which sensor type you need.  Once you do that you can use
    :meth:`getSensorPosition` and :meth:`getSensorVelocity`.  These signals
    are updated on CANBus every 20ms (by default).  If a relative sensor is
    selected, you can zero (or change the current value) using
    :meth:`setSensorPosition`.

    Analog Input and quadrature position (and velocity) are also explicitly
    reported in :meth:`getEncPosition`, :meth:`getEncVelocity`,
    :meth:`getAnalogInPosition`, :meth:`getAnalogInRaw`,
    :meth:`getAnalogInVelocity`.
    These signals are available all the time, regardless of what sensor is
    selected at a rate of 100ms.  This allows easy instrumentation for "in the
    pits" checking of all sensors regardless of modeselect.  The 100ms rate is
    overridable for teams who want to acquire sensor data for processing, not
    just instrumentation.  Or just select the sensor using
    :meth:`setFeedbackDevice` to get it at 20ms.

    Velocity is in position ticks / 100ms.

    All output units are in respect to duty cycle (throttle) which is
    -1023(full reverse) to +1023 (full forward).  This includes demand (which
    specifies duty cycle when in duty cycle mode) and rampRamp, which is in
    throttle units per 10ms (if nonzero).

[... 2082 lines cut ...]
    
    # Live Window code, only does anything if live window is activated.

    def getSmartDashboardType(self):
        return "Speed Controller"

    def updateTable(self):
        table = self.getTable()
        if table is not None:
            table.putNumber("Value", self.get())

    def valueChanged(self, itable, key, value, bln):
        self.set(float(value))

    def startLiveWindowMode(self):
        self.set(0) # Stop for safety
        super().startLiveWindowMode()

    def stopLiveWindowMode(self):
        super().stopLiveWindowMode()
        self.set(0) # Stop for safety
    
~~~

I just kinda threw in the 1000 lines of Java that need to be translated.. its going to be a pain

virtuald, 2016-01-15:
Yes, the release notice for 2016.0 mentions that the CANTalon support has not been updated yet. I would be very happy if you did it. :)

virtuald, 2016-01-15:
Also, for translating docstrings, see https://github.com/robotpy/robotpy-wpilib/blob/master/devtools/convert_javadoc.html -- it will do most of the work for you.

Arhowk, 2016-01-15:
Yeah I have about ~15 hours or so at robotics this weekend so I'll get it done in the downtimes.

I just copy pasted the java code in and replaced "public void" with "def" to start.. is the docstring thing needed? (what wouldve been nice is an update diff scanner so I can actually see which functions are new and which existed last year)

virtuald, 2016-01-15:
There is an update diff scanner thing also, but it requires validation the first time around. Checkout devtools/validate.py

You can do it by hand if you want, but the docstring thing is easier to use as it automatically converts the function definition to 'def foo' also. And I won't accept a PR without the docstrings if they exist in the Java -- our documentation should be at least as good as theirs.

virtuald, 2016-01-15:
Also, you can see what stuff is already implemented and not by using the spec scanners. Requires plyj to be installed. I haven't integrated it in very well yet, but something like this from the tests directory...

```
$ cat scanj.sh 
#!/bin/bash

cd `dirname $0`
python spec_scanners/wpilib_scanner.py ~/src/frc/allwpilib/wpilibj "$@"
```

virtuald, 2016-01-18:
Did you make progress on this?

Arhowk, 2016-01-18:
Not more than I had previously. I setup a talon SRX to test and everything
that we had previously seems to still work fine. Because of this, it
dropped in my priority list for the time being.

virtuald, 2016-02-05:
Since this has to get done at some point, I started on it too. I think the only piece left is the giant bit of logic at the end.

https://github.com/robotpy/robotpy-wpilib/tree/can2016

Arhowk, 2016-02-05:
So sorry.. I'm the only programmer (and head electrical) for my team so I tend to get wrapped up in projects and forget about everything.

commit e9922995 (2016-02-08): Update CANTalon for 2016

- Some backwards-incompatible hal_data changes, but increases consistency and should be easier to use now
- Fixes #175
commit b4e51bfb (2016-02-08): Update CANTalon for 2016

- Add unit tests for CANTalon
- Some backwards-incompatible hal_data changes, but increases consistency and should be easier to use now
- Fixes #175
commit 4b937dd1 (2016-02-08): Update CANTalon for 2016

- Add unit tests for CANTalon
- Some backwards-incompatible hal_data changes, but increases consistency and should be easier to use now
- Fixes #175
commit 3d0a2486 (2016-02-08): Update CANTalon for 2016

- Add unit tests for CANTalon
- Some backwards-incompatible hal_data changes, but increases consistency and should be easier to use now
- Fixes #175
closed 2016-02-09 by virtuald