robotpy/robotpy-wpilib #176: Exception upon deploying to robot

clxe, 2016-01-16:
When I try to deploy code to the robot with:
```sh
py robot.py deploy --builtin --nc
```
I get a `UnicodeEncodeError` thrown from the netconsole. I'm confident that this doesn't have anything to do with my code, because it's thrown when I run any of the examples as well.

Here's the contents of the console:
```
14:00:52:471 INFO    : wpilib              : WPILib version 2016.0.0
14:00:52:472 INFO    : wpilib              : HAL base version 2016.0.0; sim platform version 2016.0.0
============================= test session starts =============================
platform win32 -- Python 3.5.1, pytest-2.8.5, py-1.4.31, pluggy-0.3.1
rootdir: C:\Users\Bertucci\AppData\Local\Programs\Python\Python35\lib\site-packages\pyfrc\tests, inifile:
collected 4 items

..\..\AppData\Local\Programs\Python\Python35\lib\site-packages\pyfrc\tests\basic.py ....

========================== 4 passed in 2.40 seconds ===========================
Deploying to robot at roborio-4688-frc.local
WPILib version on robot is 2016.0.0
Using username "lvuser".
NI Linux Real-Time (run mode)

Log in with your NI-Auth credentials.

Remote working directory is /home/lvuser
mkdir /home/lvuser/py: OK
local:C:\Users\Bertucci\AppData\Local\Temp\tmpc6tni21y\py\robot.py => remote:/home/lvuser/py/robot.py
14:00:59:200 INFO    : deploy              : Netconsole is listening...

SUCCESS: Deploy was successful!
Exception in thread Thread-1:
Traceback (most recent call last):
  File "C:\Users\Bertucci\AppData\Local\Programs\Python\Python35\lib\threading.py", line 914, in _bootstrap_inner
    self.run()
  File "C:\Users\Bertucci\AppData\Local\Programs\Python\Python35\lib\threading.py", line 862, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\Bertucci\AppData\Local\Programs\Python\Python35\lib\site-packages\netconsole\netconsole.py", line 104, in run
    output_fn(do_decode(msg))
  File "C:\Users\Bertucci\AppData\Local\Programs\Python\Python35\lib\site-packages\netconsole\netconsole.py", line 20, in _output_fn
    sys.stdout.write(s)
  File "C:\Users\Bertucci\AppData\Local\Programs\Python\Python35\lib\encodings\cp437.py", line 19, in encode
    return codecs.charmap_encode(input,self.errors,encoding_map)[0]
UnicodeEncodeError: 'charmap' codec can't encode character '\u2794' in position 0: character maps to <undefined>
```

virtuald, 2016-01-17:
Very interesting, what locale do you have set on your computer?

virtuald, 2016-01-17:
Also, it occurs to me -- are you on Windows 10? I feel like something about the console changed there, and I haven't tested on Windows 10 lately.

clxe, 2016-01-17:
Yes, I'm using Windows 10. The locale is set to US English.

PeterJohnson, 2016-01-17:
Sometimes netconsole can spit out a bit of junk at the initial connection (you can see this in the Driver Station log as well).  I would recommend the python netconsole catch this exception and fall back to doing something like displaying as escaped characters instead.

virtuald, 2016-01-18:
This is a Windows 10/Python 3.5 specific issue, I was able to duplicate it on a Windows 10 laptop. I think the correct answer is to have python escape undecodable characters, I'll try to get this fixed tonight.

virtuald, 2016-01-18:
Fixed, pushed to pypi. You can get the fix by doing `py -3 -m pip install -U pynetconsole`. I'll update pyfrc soon too.

closed 2016-01-18 by virtuald