robotpy/robotpy-wpilib #177: _i2CTransaction function failing leading to inability to use i2c bus

ArthurAllshire, 2016-01-20:
I2C reading and writing does not work on the roboRIO. It seems like the fail to _i2CTransaction in i2CTransaction fails.
Code:
    from wpilib import I2C

    class BoschGyro():
        def __init__(self):
            self.gyro = I2C(I2C.Port.kMXP, 0x29)
    
        def exists(self):
            return self.gyro.addressOnly()
Fails on call to addressOnly()
Error I am getting on the driver station:

    File "/home/lvuser/py/robot.py", line 84, in teleopInit
        self.logger.info("Gyro: ", self.gyro.exists())
    File "/home/lvuser/py/subsystems/bosch_gyro.py", line 8, in exists
        return self.gyro.addressOnly()
    File "/usr/local/lib/python3.5/site-packages/wpilib/i2c.py", line 74, in addressOnly
        self.transaction([], 0)
    File "/usr/local/lib/python3.5/site-packages/wpilib/i2c.py", line 63, in transaction
        dataToSend, receiveSize)
    File "/usr/local/lib/python3.5/site-packages/hal/functions.py", line 402, in i2CTransaction
        rv = _i2CTransaction(port, device_address, send_buffer, send_size, recv_buffer, receive_size)
    ctypes.ArgumentError: argument 5: <class 'TypeError'>: expected LP_c_ubyte instance instead of _ctypes.PyCArrayType

I'm just calling the function from teleopInit(), and it completely crashes the robot code.

virtuald, 2016-01-20:
This problem is fixed in the latest master, I just haven't made a release yet (you should subscribe to the robotpy mailing list :) ).

Master also contains functionality that allows you to simulate i2c devices, check out https://github.com/robotpy/robotpy-wpilib/blob/master/wpilib/tests/test_i2c.py. I was hoping to add a simulator for one of the gyros as an example before making the release.

ArthurAllshire, 2016-01-21:
@virtuald How would I go about installing the current master to the roborio or should we just wait until the next release?

Thanks for responding so quickly. Definitely signed up to the mailing list ;)

virtuald, 2016-01-21:
See `devtools/build_and_deploy.sh`. I'm trying to get it out tonight (so.. maybe in the next 4 hours).

virtuald, 2016-01-21:
Pushed 2016.1.0 to pypi. Added more stuff to the i2c/spi simulators, there will be an update of pyfrc to support some of it too.

closed 2016-01-21 by virtuald