robotpy/robotpy-wpilib #178: Client Connected with bad protocol revision

ariovistus, 2016-01-23:
When trying to run last year's code on last year's robot with this year's robotpy and this year's driver station, I'm seeing 

Client Connected with bad protocol revision: 0x300

and that's coming from pynetworktables which does not appear to have a release for this year. Is that the case currently?

PeterJohnson, 2016-01-23:
Correct, pynetworktables does not yet support NetworkTables 3.0.  Dashboards (e.g. SmartDashboard) that use the ntcore library will automatically reconnect with the 2.0 protocol.  However, the LabView dashboard will not (this year's dashboard only supports the 3.0 protocol).

virtuald, 2016-01-29:
Duplicate of https://github.com/robotpy/pynetworktables/issues/21

closed 2016-01-29 by virtuald