robotpy/robotpy-wpilib #183: OSError: dlopen(libnivision.so, 6): image not found

codetheweb, 2016-02-16:
I'm getting a weird error when doing `python3 robot.py deploy`.  I've already tried upgrading `pyfrc` with `pip3 install pyfrc --upgrade`.
```
Traceback (most recent call last):
  File "robot.py", line 2, in <module>
    import wpilib
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/wpilib/__init__.py", line 86, in <module>
    from .cameraserver import *
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/wpilib/cameraserver.py", line 2, in <modu
le>
    import nivision
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/nivision/__init__.py", line 7, in <module
>

    from . import camera
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/nivision/camera.py", line 1, in <module>
    from . import private
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/nivision/private.py", line 6, in <module>
    from . import core
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/nivision/core.py", line 16, in <module>
    _dll = ctypes.CDLL("libnivision.so")
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/ctypes/__init__.py", line 347, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: dlopen(libnivision.so, 6): image not found
```

The robot code is fairly simple:

```
import wpilib

class MyRobot(wpilib.IterativeRobot):

    def robotInit(self):
        self.stick = wpilib.Joystick(1)
        self.arduino = wpilib.I2C(wpilib.I2C.Port.kOnBoard, 4)

    def teleopPeriodic(self):
        if (self.stick.getTrigger()):
            self.arduino.transaction(b'go', 0)

if __name__ == "__main__":
    wpilib.run(MyRobot)
```

virtuald, 2016-02-16:
I'm not sure why you have nivision installed locally... that will only work on the robot. You should do a 'pip3 uninstall pynivision' and that should clear up your issue.

codetheweb, 2016-02-17:
Thanks, that fixed it!  No idea how that got installed.

closed 2016-02-17 by codetheweb