robotpy/robotpy-wpilib #202: Traceback Issue wpilib.run(MyRobot)

agentHP007, 2016-10-20:
So while attempting to deploy any code to the roborio was giving me this error even after updating all the packages and everything 

`
14:50:17:493 INFO    : wpilib              : WPILib version 2016.2.4
14:50:17:493 INFO    : wpilib              : HAL base version 2016.2.4; sim platform version 2016.2.4
============================= test session starts =============================
platform win32 -- Python 3.5.2, pytest-3.0.3, py-1.4.31, pluggy-0.4.0
rootdir: C:\Users\sphstech\Desktop\PyDev\Getting Started\tests, inifile: 
collected 5 items

pyfrc_test.py .....

========================== 5 passed in 0.84 seconds ===========================
Robot setup (hit enter for default value):
Robot hostname (like roborio-XXX-frc.local, or an IP address): ERASED FOR PRIVACY REASONS
Looking up hostname ERASED FOR PRIVACY REASONS ...
Traceback (most recent call last):
  File "C:\Users\sphstech\Desktop\PyDev\Getting Started\src\robot.py", line 41, in <module>
    wpilib.run(MyRobot)
  File "C:\Users\sphstech\AppData\Local\Programs\Python\Python35-32\lib\site-packages\wpilib\_impl\main.py", line 101, in run
    retval = options.cmdobj.run(options, robot_class, **kwargs)
  File "C:\Users\sphstech\AppData\Local\Programs\Python\Python35-32\lib\site-packages\pyfrc\mains\cli_deploy.py", line 159, in run
    no_resolve=options.no_resolve)
  File "C:\Users\sphstech\AppData\Local\Programs\Python\Python35-32\lib\site-packages\pyfrc\robotpy\installer.py", line 333, in ssh_from_cfg
    for _, t, _, _, (ip, _) in addrs:
ValueError: too many values to unpack (expected 2)`

this was in an attempt to the deploy just the Get Started Example code
What can i do to solve this error?

--Thank You For Your Time

virtuald, 2016-10-21:
Most likely the IP address of the RoboRIO was an IPv6 address.This was fixed in https://github.com/robotpy/pyfrc/issues/56, but I hadn't pushed an updated release of pyfrc to pypi. I just pushed a version with that fix in it -- upgrade pyfrc (see the docs) and let me know how it goes.

virtuald, 2016-10-21:
Sorry, fixed in https://github.com/robotpy/pyfrc/pull/57.

closed 2016-10-21 by virtuald