robotpy/robotpy-wpilib #205: Use flat namespace for hal_data instead of nested dictionaries

virtuald, 2016-11-02:
I'm sure there was a reason I didn't do this before... but basically, right now we access the data like so:

```
hal_data['pdp']['has_source']
hal_data['solenoid'][0]['value']
... etc
```

I want to change it to this instead:

```
hal_data['pdp.has_source']
hal_data['solenoid.0.value']
```

This will make a few things easier to do, in particular it'll be significantly easier to create diffs of the hal_data if you maintained separate sets of dictionaries for I/O, as it would just be `dict.update()` to copy the updates over. This would be useful if trying to ship the data over a websocket, as you would only need to send changes.

There also would be fewer dictionary lookups in the simulator execution, which is probably a good thing. We could cache a lot of the keys in the fake port objects and such so that we're not constantly computing 

For backwards compatibility, we could define a dictionary shim that would map the existing dictionaries to the new keys (and this might be useful for teams writing custom physics... ).

If you have opinions, state them here. I'm not 100% sold on it yet.

virtuald, 2016-11-02:
@Twinters007 @computer-whisperer may have opinions on this?

Twinters007, 2016-11-05:
I like it. Seems like a good change. What's keeping you from being 100% sure?

virtuald, 2017-01-06:
This will have to wait until 2018.

virtuald, 2017-11-03:
Time to revisit this... I had thought of a reason to not do it, but I've forgotten it.

Twinters007, 2017-11-03:
Could it be accessing dicts based on variables will be easier?

twinters007, 2018-01-06:
```
def __getattr__(self, attr):
        return self.get(attr)

def __setattr__(self, key, value):
    self.__setitem__(key, value)
```
Would this simple solution work for hal?

auscompgeek, 2018-01-06:
I think this will have to wait until next year. It's a bit late to be thinking of architectural changes now.

virtuald, 2018-12-07:
Hm. Is now the time for change?

virtuald, 2018-12-07:
Hm, I think the reason to not do it is because NotifyDict adds overhead and it would have to be added at the top level... but honestly, I don't know that anything actually uses it?

ArchdukeTim, 2018-12-20:
https://docs.python.org/3/library/types.html#types.SimpleNamespace

could this work?

virtuald, 2019-01-04:
Sad, will have to push this off to 2020 methinks.

virtuald, 2020-01-19:
OBE. Maybe.

closed 2020-01-19 by virtuald