robotpy/robotpy-wpilib #208: Bring 2017 WPILib changes to RobotPy

virtuald, 2016-11-14:
While there will still be some churn, a big majority of breaking changes have been pushed to WPILib and so the RobotPy project needs to bring them in before kickoff. This is a meta-bug that I'll use to track the various pieces of things as they get done.

This depends on #207 , as I expect most of the changes will be dependent on the HAL interface changes... I'm going to take a look at this tonight and see if I can bang the HAL stuff out.

Once the HAL stuff is done, I suspect the rest will be relatively easy. In particular, using git-source-track it should be fairly straightforward to figure out what files changed and what the changes are. See the documentation I created at http://robotpy.readthedocs.io/en/latest/internal.html#updating-robotpy-source-code-to-match-wpilib

One problem is that actual hardware testing of the 2017 changes will require HAL changes, which will require the 2017 beta images. Anyone who doesn't have access to the beta test images will run into problems trying to test it there -- but, as you know the best part about RobotPy is that most of the testing should be doable using the simulated HAL support. If anyone who makes significant contributions wants to test on real hardware, contact me directly.

If there are people interested in helping with this, I'd be greatly appreciative! @james-ward @FRC2539 @amorygalili or anyone else...

bb20basketball, 2016-11-14:
I honestly have no idea how helpful I would be, but if there is something that I could do, I will gladly help out to move RobotPy to the next version.

virtuald, 2016-11-14:
@bb20basketball one thing that would be useful that you could do -- test stuff! In particular, I would love for people to use the [new alpha version of pynetworktables](https://robotpy.github.io/2016/10/28/pynetworktables-nt3-rewrite/) and let me know if they run into problems with it.

ryannazaretian, 2016-11-16:
I'm in.

virtuald, 2016-11-16:
@ryannazaretian awesome, welcome! @auscompgeek @ArthurAllshire are helping out too.

I think the best approach right now is to just identify a file you want to work on, and let others know that you're working on it. There are some files that don't depend on HAL changes, so those are a great first place to start. Otherwise, until the HAL changes are merged in, the tests will break, but that's fine. And feel free to use HAL functions that don't exist yet... they will eventually!

I believe @ArthurAllshire was looking at SmartDashboard?

I have barely started on the HAL changes. I'm hoping to get it done soon.

ArthurAllshire, 2016-11-16:
@virtuald yes I'm working on the smartdashboard stuff. I'm about 1/2 way through the changes at the moment.

virtuald, 2016-11-17:
Note the naming convention for HAL stuff has changed slightly, as mentioned in #207.

virtuald, 2016-11-22:
I think the HAL function signatures are mostly correct at this point in my branch. I'm going to update the DS class so that basic control works and see how that goes... the actual HAL simulated implementation is still rather incorrect however, but that will have to come later.

virtuald, 2016-11-24:
The DriverStation code is done I think (with the exception of reportError), I'm going to continue on the task of getting simulation to work by converting robotBase et al

virtuald, 2016-11-24:
Ok, I've merged the hal branch into master, along with the DriverStation changes. I can start a basic program that doesn't create any objects and doesn't do anything now! :)

Tomorrow I'm going to start working on other classes, and I'll be updating the HAL as I fix things. If you port other objects, feel free to update their HAL implementation.

There are minor updates in pyfrc required too.

virtuald, 2016-11-26:
I updated the HAL scanner tests instead. Tomorrow, I'll start working on other objects. If you're working on something, post here! I'll do the same.

virtuald, 2016-11-26:
I'm going to update the PWM + related classes now.

virtuald, 2016-11-27:
The PWM related changes are done, so I'm done for tonight. The simple example now works in simulation. :)

ArthurAllshire, 2016-11-27:
@virtuald I havent ended up doing the error reporting stuff with the driver station, as I am a bit out of my depth with it...

virtuald, 2016-11-27:
No worries, there's lots of other things that still need to be ported. :)

ArthurAllshire, 2016-11-29:
ill work on the generichid stuff (joystick/gamepad) as long as nobody else is.

Twinters007, 2016-12-04:
I'm working on things that start with the letter 'a'

ryannazaretian, 2016-12-04:
Dustin,

Very sorry. With packing to move and still trying to mentor my team, I
don't have the time for porting changes.

Thanks,
Ryan Nazaretian

virtuald, 2016-12-05:
@ryannazaretian no worries

virtuald, 2016-12-05:
This weekend, I've been working on figuring out how robotpy-ctre is going to work (I think I'm mostly there...), upgrading my roborio et al to the 2017 tools, upgrading the roborio-vm to use a new kernel, and starting to build 2017 ipk packages.

virtuald, 2017-01-07:
This is good to go, thanks everyone who helped!

closed 2017-01-07 by virtuald