robotpy/robotpy-wpilib #212: Extract CANTalon to robotpy-ctre package

virtuald, 2016-11-15:
Related issues:

* #206
* #208

Package management in python is trivial, so I think it makes sense for these types of things to be separate packages, despite that the RobotPy project will be maintaining them. New repo will be at https://github.com/robotpy/robotpy-ctre

virtuald, 2016-11-21:
Removed CANTalon in ed58c9e67e334057d6416e6de6859f7f6b549ffe

vanjac, 2016-11-30:
Maybe it would be good to have a placeholder CANTalon class that prints a deprecation warning when methods are called, to prevent initial confusion when teams update to the new changes?

virtuald, 2016-11-30:
I could add the following:

```
class CANTalon:
    def __init__(self):
        raise ValueError("CANTalon has moved to ctre.CANTalon, you must install robotpy-ctre")
```

vanjac, 2016-11-30:
That would work. __init__ could allow *args and **kwargs so it would ignore any arguments.

virtuald, 2017-01-07:
CANTalon support will be delayed, hopefully get it done soon.

vanjac, 2017-01-07:
No way to use CANTalons at all yet?

virtuald, 2017-01-07:
Yeah, lots of other things had to be done, and there's only so much time to do it. It is my number one priority for this weekend (after reading the rules for the game).

vanjac, 2017-01-07:
Will tests for robot code with talons still pass?

virtuald, 2017-01-07:
The PWM interface for Talon and TalonSRX are still there. Only the CAN version of the motor controller is affected by this.

virtuald, 2017-01-10:
I've got something working, and I think I might have a full solution available tomorrow night.

virtuald, 2017-01-13:
Released last night.

closed 2017-01-13 by virtuald