robotpy/robotpy-wpilib #243: Python 3.6 RoboRIO startup issue

virtuald, 2017-01-07:
```
BlockingIOError: [Errno 11] Resource temporarily unavailable 
  
 The above exception was the direct cause of the following exception: 
  
 Traceback (most recent call last): 
   File "/home/lvuser/py/robot.py", line 15, in <module> 
     import wpilib 
   File "/usr/local/lib/python3.6/site-packages/wpilib/__init__.py", line 72, in <module> 
     from ._impl.main import run 
   File "/usr/local/lib/python3.6/site-packages/wpilib/_impl/main.py", line 7, in <module> 
     from pkg_resources import iter_entry_points 
   File "/usr/local/lib/python3.6/site-packages/pkg_resources/__init__.py", line 36, in <module> 
     import email.parser 
   File "/usr/local/lib/python3.6/email/parser.py", line 12, in <module> 
     from email.feedparser import FeedParser, BytesFeedParser 
   File "/usr/local/lib/python3.6/email/feedparser.py", line 27, in <module> 
     from email._policybase import compat32 
   File "/usr/local/lib/python3.6/email/_policybase.py", line 9, in <module> 
     from email.utils import _has_surrogates 
   File "/usr/local/lib/python3.6/email/utils.py", line 28, in <module> 
     import random 
   File "/usr/local/lib/python3.6/random.py", line 742, in <module> 
     _inst = Random() 
 SystemError: <class 'random.Random'> returned a result with an error set 
```

This seems to be caused by running

https://www.python.org/dev/peps/pep-0522/#raising-blockingioerror-in-os-urandom-on-linux

The behavior is that the robot program restarts a few times, but eventually gets fixed and the program starts up. It's weird, because it shouldn't be using urandom at this point... will need to investigate this and fix it before the end of the season.

virtuald, 2017-01-08:
Bug filed on python at http://bugs.python.org/issue29208

virtuald, 2017-01-08:
@auscompgeek thinks that https://github.com/python/cpython/commit/ff4a0794fe5e2cf891f9ee3cf33343c77881fe38 would fix this.

virtuald, 2017-01-10:
I'm running a build now with the patch, so hopefully will be able to test this later tonight.

virtuald, 2017-01-10:
Fixed in Python 3.6.0-r1 in the ipkg feed

closed 2017-01-10 by virtuald

auscompgeek, 2017-03-04:
In case anyone's looking at this and is curious as to where the commit is in the new cpython tree: python/cpython@acc2f74ca991f6579f5b9055dfe885b9a8af4645