robotpy/robotpy-wpilib #245: Code is not persistent during roboRIO restarts

CadeBrown, 2017-01-09:
Any code deployed to the roboRIO does not persist when restarting roboRIO.

i.e. on the driver station, the Robot Code indicator is red instead of green, and whenever you try to run, the screen flashes and says "No Robot Code".

The roboRIO has been imaged with the 2017 image, and the driver station is a fresh install of windows with the 2017 FRC official driver station.

Code works fine for as many times as you run or switch modes, but when turning the roboRIO on and then back off, it displays as no robot code.

virtuald, 2017-01-10:
Interesting. Did you open the Driver Station console to see if there's anything there? You might be running into https://github.com/robotpy/robotpy-wpilib/issues/243

CadeBrown, 2017-01-10:
Actually, it eventually showed up green after about  30 minutes of leaving it open. Now, it still takes 15-20 seconds to show green robot code, everytime the roboRIO is restarted.

virtuald, 2017-01-10:
So, I just tried this on my RoboRIO, and it's constantly restarting code over and over and over with the random.Random error. Previously it had only occurred 3 times and then it started. There is a proposed fix on the python bug tracker, so I'm going to try that soon and it should resolve the issue. Going to close this unless you can confirm it's a different bug.

closed 2017-01-10 by virtuald