robotpy/robotpy-wpilib #246: installation error on Windows

vScourge, 2017-01-10:
Following the installation instructions, I installed Python 3.5.1 (32-bit), downloaded the robotpy 2017.0.0 zipfile, unzipped it, and ran this command from that folder:

py -3 installer.py install-robotpy

It fails with this exception:

```
C:\Users\User\Desktop\robotpy-2017.0.0>py -3 installer.py install-robotpy
Traceback (most recent call last):
  File "installer.py", line 1010, in <module>
    retval = main()
  File "installer.py", line 991, in main
    retval = options.cmdobj(options)
  File "installer.py", line 751, in install_robotpy
    self.install_opkg(opkg_options)
  File "installer.py", line 798, in install_opkg
    opkg = self._get_opkg()
  File "installer.py", line 676, in _get_opkg
    opkg.add_feed("http://download.ni.com/ni-linux-rt/feeds/2016/arm/ipk/cortexa9-vfpv3")
  File "installer.py", line 93, in add_feed
    self.load_package_db(feed)
  File "installer.py", line 107, in load_package_db
    for line in fp.readlines():
  File "C:\Users\User\AppData\Local\Programs\Python\Python35-32\lib\encodings\cp1252.py", line 23, in decode
    return codecs.charmap_decode(input,self.errors,decoding_table)[0]
UnicodeDecodeError: 'charmap' codec can't decode byte 0x9d in position 7516: character maps to <undefined>
```

This is on Windows 10, on a PC that fomerly had last year's robotpy successfully installed under Python 3.4.  I uninstalled that before doing the above.

virtuald, 2017-01-10:
Stupid Windows. Give me a minute, I'll take a look.

virtuald, 2017-01-10:
Sorry about that, I don't use Windows very often. I'm going to hold off to do a release until I resolve #243, but you can get the installer directly from git if you're impatient. There's also a fix required for pyfrc too, so I'll update that shortly.

closed 2017-01-10 by virtuald

vScourge, 2017-01-10:
That was fast.  Thanks, Dustin, we'll give it a spin tomorrow.

commit 29c496cd (2017-01-10): installer: fix encoding issue on Windows

- Fixes #246