robotpy/robotpy-wpilib #247: Failing to install on roborio

vScourge, 2017-01-11:
I'm getting the following errors when installing robotpy to the roborio.  I'm using the automated installation, which worked fine for us last season.  This is on Windows 10.

`C:\Users\User\Desktop\robotpy-2017>py installer.py install-robotpy
Copying over the opkg cache...
Looking up hostname 10.40.96.2 ...
Connecting to robot via SSH at 10.40.96.2
Copying over the pip cache...
Using username "admin".
NI Linux Real-Time (run mode)

Log in with your NI-Auth credentials.

Remote working directory is /home/admin
local:C:\Users\User\Desktop\robotpy-2017\pip_cache\pynetworktables-2017.0.1.tar.gz => remote:/home/admin/pip_cache/pynetworktables-2017.0.1.tar.gz
Running installation...
Done.
Installing python36 (3.6.0-r1) on root.
Collected errors:
 * check_data_file_clashes: Package python36 wants to install file /usr/local/lib/libpython3.so
        But that file is already provided by package  * python3
 * check_data_file_clashes: Package python36 wants to install file /usr/local/bin/pydoc3
        But that file is already provided by package  * python3
 * check_data_file_clashes: Package python36 wants to install file /usr/local/bin/python3
        But that file is already provided by package  * python3
 * check_data_file_clashes: Package python36 wants to install file /usr/local/bin/pyvenv
        But that file is already provided by package  * python3
 * check_data_file_clashes: Package python36 wants to install file /usr/local/bin/2to3
        But that file is already provided by package  * python3
 * check_data_file_clashes: Package python36 wants to install file /usr/local/bin/pip3
        But that file is already provided by package  * python3
 * check_data_file_clashes: Package python36 wants to install file /usr/local/bin/idle3
        But that file is already provided by package  * python3
 * opkg_install_cmd: Cannot install package python36.
Command '['C:\\Users\\User\\Desktop\\robotpy-2017\\win32\\plink.exe', '-pw', '', 'admin@10.40.96.2', 'bash opkg_cache/install_opkg.sh && /usr/local/bin/pip3 install --no-index --find-links=pip_cache --pre --upgrade wpilib robotpy-hal-base robotpy-hal-roborio robotpy-wpilib-utilities']' returned non-zero exit status 255`

virtuald, 2017-01-11:
Did you reimage your roborio for 2017? It sounds like you have last year's stuff installed.

vScourge, 2017-01-11:
facepalm.  Thank you.

closed 2017-01-11 by virtuald