robotpy/robotpy-wpilib #251: For IterativeRobots, teleopPeriodic is running ~720 times per second

vanjac, 2017-01-12:
I wrote this to test the timing for `teleopPeriodic`:
```
import wpilib

class TestRobot(wpilib.IterativeRobot):
    
    def robotInit(self):
        self.count = 0

    def teleopPeriodic(self):
        if self.count % 50 == 0:
            print("1 second")
        self.count += 1

if __name__ == "__main__":
    wpilib.run(TestRobot)
```

In 5 seconds, "1 second" was printed 72 times. So it seems like `teleopPeriodic` is being called ~720 times per second, instead of ~50 according to the documentation.

To make sure it wasn't Driver Station sending packets too quickly, I also wrote this:
```
import wpilib

class TestRobot(wpilib.IterativeRobot):
    
    def robotInit(self):
        self.joystick = wpilib.Joystick(0)

    def teleopPeriodic(self):
        xPos = self.joystick.getX()
        print(xPos)

if __name__ == "__main__":
    wpilib.run(TestRobot)
```
And ran it while moving the joystick quickly. The rate didn't slow and same values were printed repeatedly. So maybe it just isn't properly waiting for a new packet before running `teleopPeriodic` again.

virtuald, 2017-01-12:
Hm, that's definitely a problem. All of the waiting stuff is in the DriverStation class... there's a thread that calls [hal.waitForDSData](https://github.com/robotpy/robotpy-wpilib/blob/master/wpilib/wpilib/driverstation.py#L619), maybe it's returning too quickly?

Oh ha, self.waitForDataPredicate isn't being reset.

Twinters007, 2017-01-13:
Out of curiosity, what made you question the timing?

b-mck, 2017-01-13:
@Twinters007 We were actually testing timing for some dead-reckoning code. To further test we logged joystick input in TeleopPeriodic and saw dozens of identical updates where there should have been one.

vanjac, 2017-01-13:
I can work on a pull request for this

virtuald, 2017-01-13:
I have a fix, pushing it now actually.

commit 50e172a5 (2017-01-13): DriverStation: waitForData wasn't waiting for data

- Fixes #251
closed 2017-01-13 by virtuald