robotpy/robotpy-wpilib #252: hal.exceptions.HALError: HAL: No available resources to allocate

vScourge, 2017-01-14:
We're getting this error on two different robots, different codebases, both using robotpy 2017.0.3.  It only happens if there's a wpilib.Solenoid object created.  One robot did not have this problem with the last robotpy 2016 libraries, and the hardware all checks out on both.

```
Error at /usr/local/lib/python3.6/site-packages/wpilib/_impl/main.py.114:run: ? Launching «'/usr/local/bin/python3' '-u' '-O' '/home/lvuser/py/robot.py' 'run'»
14:58:35:196 INFO    : wpilib              : WPILib version 2017.0.1
14:58:35:200 INFO    : wpilib              : HAL base version 2017.0.1; roboRIO platform version 2017.0.1
14:58:35:201 INFO    : wpilib              : HAL library version 2017.1.1
14:58:35:211 INFO    : nt                  : NetworkTables 2017.0.1 initialized in server mode
14:58:35:273 INFO    : nt                  : CONNECTED 10.40.96.55 port 41367 ()
14:58:40:345 INFO    : wpilib.PIDTask0     : Timer task started
14:58:40:357 WARNING : robotpy             : Robots don't quit!
14:58:40:380 ERROR   : wpilib.ds           : ERROR Unhandled exception
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/wpilib/robotbase.py", line 178, in main
    robot.startCompetition()
  File "/usr/local/lib/python3.6/site-packages/wpilib/iterativerobot.py", line 78, in startCompetition
    self.robotInit()
  File "/home/lvuser/py/robot.py", line 87, in robotInit
    self.drive = subsystems.drivetrain.Drivetrain( self )
  File "/home/lvuser/py/subsystems/drivetrain.py", line 70, in __init__
    self.shift_solenoid_1 = wpilib.Solenoid( const.ID_DRIVE_SOLENOID_SHIFTER_1 )
  File "/usr/local/lib/python3.6/site-packages/wpilib/solenoid.py", line 80, in __init__
    self._solenoidHandle = hal.initializeSolenoidPort(portHandle)
  File "/usr/local/lib/python3.6/site-packages/hal/functions.py", line 42, in outer
    raise HALError(getErrorMessage(status.value))
hal.exceptions.HALError: HAL: No available resources to allocate
```

virtuald, 2017-01-15:
Hm, I haven't tried solenoids yet this year. Did you only create a single solenoid? Multiple solenoids? If you could paste your initialization code that could be helpful.

vScourge, 2017-01-15:
We started with four solenoids, port numbers 0-3.  Then we cut it down to just one on port zero and it still puts up this error.  Here's the whole drivetrain init function.  The constant for that solenoid is set to 0.

```
DEFAULT_KP = 0.02
DEFAULT_KI = 0.01
DEFAULT_KD = 0.1

class Drivetrain( wpilib.command.PIDSubsystem ):

	def __init__( self, robot ):

		super( ).__init__( DEFAULT_KP, DEFAULT_KI, DEFAULT_KD, name = 'drive' )

		self.robot = robot

		self.drive = wpilib.RobotDrive( frontLeftMotor = const.ID_DRIVE_MOTOR_FRONT_LEFT,
		                                rearLeftMotor = const.ID_DRIVE_MOTOR_REAR_LEFT,
										frontRightMotor = const.ID_DRIVE_MOTOR_FRONT_RIGHT,
										rearRightMotor = const.ID_DRIVE_MOTOR_REAR_RIGHT )

		self.strafe_wheel = wpilib.VictorSP( const.ID_STRAFE_WHEEL )
		self.drive_state = const.ID_H_DRIVE
		
		self.shift_solenoid_1 = wpilib.Solenoid( const.ID_DRIVE_SOLENOID_SHIFTER_1 )
		#self.shift_solenoid_2 = wpilib.Solenoid( const.ID_DRIVE_SOLENOID_SHIFTER_2 )
		#self.actuate_solenoid_1 = wpilib.Solenoid( const.ID_DRIVE_SOLENOID_ACTUATOR_1 )
		#self.actuate_solenoid_2 = wpilib.Solenoid( const.ID_DRIVE_SOLENOID_ACTUATOR_2 )
```

virtuald, 2017-01-15:
The validation for Solenoid/SolenoidBase was done incorrectly, it wasn't updated to reflect the 2017 WPILib. I'll address it tonight.

closed 2017-01-15 by virtuald

virtuald, 2017-01-15:
Fixed and released as 2017.0.4

commit 6ecf7d09 (2017-01-15): Update solenoid implementation to comply with 2017 changes

- Fixes #252