robotpy/robotpy-wpilib #259: Can not initialize an instance of the Encoder class

saketramdurbha, 2017-02-04:
I have not been able to initialize an instance of the `Encoder` class at all. 

Here is my simplified robot code that I have been using:

    import wpilib
    from wpilib import Encoder, IterativeRobot

    class MyRobot(IterativeRobot):

        def robotInit(self):
            self.encoderFrontLeft = Encoder(0, 1)

        def teleopPeriodic(self):
            print("Encoder Front Left Distance: " + str(self.encoderFrontLeft.getDistance()))

    if __name__ == "__main__":
        wpilib.run(MyRobot)

Here is the logfile: https://gist.github.com/saketramdurbha/7a196b3506d1d3a1bf1ec630c776c99b#file-logfile-txt

virtuald, 2017-02-04:
Sorry about this, this bug occurs because the simulator doesn't validate return types... I always wondered if that would come back to bite me. I'll get a fix out tonight.

closed 2017-02-05 by virtuald

virtuald, 2017-02-05:
This bug is fixed in the latest release of RobotPy. Let me know if you have more issues.

commit e15b152c (2017-02-05): Fix encoder to match Java

- Fixes #259