robotpy/robotpy-wpilib #261: Digital Output Attribute Error

codetheweb, 2017-02-15:
I'm trying to use a digital output.  When setting the value with `output.set(True)`, I'm getting this error on upload:
```
self = <wpilib.digitaloutput.DigitalOutput object at 0x10e9cea20>, value = True

    def set(self, value):
        """Set the value of a digital output.

            :param value: True is on, off is False
            :type  value: bool

            """
>       hal.setDIO(self.port, 1 if value else 0)
E       AttributeError: 'DigitalOutput' object has no attribute 'port'
```

I looked at the object; and it indeed doesn't appear to have a `port` attribute.  Any ideas on what's going on?

virtuald, 2017-02-15:
Means someone done messed up. I'll upload a fix in a few minutes.

virtuald, 2017-02-15:
Amusingly, there was a unit test to cover DigitalOutput.get, but not set. :)

closed 2017-02-15 by virtuald

virtuald, 2017-02-15:
Latest release uploaded to pypi and github. Let me know if you run into other issues!

codetheweb, 2017-02-15:
Now working, thanks!

commit 26c45b20 (2017-02-15): Fix DigitalOutput.set

- Add a test to cover it
- Fixes #261