robotpy/robotpy-wpilib #262: Sonar class HAL error

czhao39, 2017-02-15:
When I instantiate a `robotpy_ext.common_drivers.xl_max_sonar_ez.MaxSonarEZPulseWidth` object, I get the following error:

> ..\..\venv\lib\site-packages\pyfrc\tests\basic.py:32:
\_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
..\..\venv\lib\site-packages\pyfrc\test_support\controller.py:148: in run_test
    self._robot.startCompetition()
..\..\venv\lib\site-packages\commandbased\commandbasedrobot.py:24: in startCompetition
    self.robotInit()
robot/robot.py:48: in robotInit
    ???
..\inputs\sonar.py:15: in init
    front = MaxSonarEZPulseWidth(robotmap.sonar.front_channel, meter)
..\..\venv\lib\site-packages\robotpy_ext\common_drivers\xl_max_sonar_ez.py:32: in __init__
    self.counter = wpilib.Counter(channel)
..\..\venv\lib\site-packages\wpilib\counter.py:151: in __init__
    self.setUpSource(upSource)
..\..\venv\lib\site-packages\wpilib\counter.py:238: in setUpSource
    source = DigitalInput(channel)
..\..\venv\lib\site-packages\wpilib\digitalinput.py:35: in __init__
    super().__init__(channel, True)
..\..\venv\lib\site-packages\wpilib\digitalsource.py:45: in __init__
    self._handle = hal.initializeDIOPort(hal.getPort(channel), input)
\_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
>
>args = (<PortHandle at 0x4eb23b0 pin=0 module=0>, True), kwargs = {}
>status = c_long(-1029), rv = None
>
>    def outer(*args, **kwargs):
        status = C.c_int32(0)
        rv = _inner(*args, status=status, **kwargs)
        if status.value == 0:
            return rv
        elif status.value < 0:
>           raise HALError(getErrorMessage(status.value))
E           hal.exceptions.HALError: HAL: A resource is already allocated.
>
>..\..\venv\lib\site-packages\hal\functions.py:42: HALError

virtuald, 2017-02-15:
What other objects are you creating?

czhao39, 2017-02-15:
NavX and USB camera

czhao39, 2017-02-15:
Oh shoot I'm an idiot. I had a switch on the same port that I'd made for testing. Sorry (again).

closed 2017-02-15 by czhao39

virtuald, 2017-02-15:
Not a problem.