robotpy/robotpy-wpilib #264: SendableChooser fails to get selection when not on field

vScourge, 2017-03-28:
This year we're using SendableChooser again to select an autonomous mode from the SmartDashboard.  This year in our shop environment (wifi) it is failing to use the selected auto mode.  Oddly, this worked fine at competition last week when connected to the field.  

Looking at the code, it appears that the following call in sendablechooser.py, line 90 in getSelected() is not working as expected:

`table = self.getTable()`

The "getTable" call always returns None, which causes it to use the default mode, regardless of the selection.  If no default mode/object is defined it will throw an exception trying to look up "None" in the map.

This is using pyfrc 2017.1.3, robotpy 2017.1.2, on Windows 10.

virtuald, 2017-03-28:
What code are you using to test this?

It appears that table gets set via [initTable](https://github.com/robotpy/robotpy-wpilib/blob/master/wpilib/wpilib/sendablechooser.py#L99), which gets called when you call [SmartDashboard.putData](https://github.com/robotpy/robotpy-wpilib/blob/master/wpilib/wpilib/smartdashboard.py#L87).

vScourge, 2017-03-28:
In "oi.py" we have this:

```
		self.auto_choose = SendableChooser( )
	
		self.auto_choose.addObject( 'Do Nothing', commands.autonomous.Do_Nothing( self.robot ) )
		self.auto_choose.addObject( 'Center Gear', commands.autonomous.Move_To_Gear( self.robot, const.ID_AUTO_CENTER_GEAR ) )
		self.auto_choose.addObject( 'Right Gear', commands.autonomous.Move_To_Gear( self.robot, const.ID_AUTO_RIGHT_GEAR ) )
		self.auto_choose.addObject( 'Left Gear', commands.autonomous.Move_To_Gear( self.robot, const.ID_AUTO_LEFT_GEAR ) )
		self.auto_choose.addObject( 'No Camera Center', commands.autonomous.Center_Gear_Without_Camera( self.robot ) )
		self.auto_choose.addDefault( 'Boiler Red', commands.autonomous.Hit_Hopper( self.robot, const.ID_AUTO_RED_SIDE ) )	
	
		SmartDashboard.putData( 'Autonomous Mode', self.auto_choose )
```

Then in "robot.py", in our IterativeRobot subclass there's this:

```
	def autonomousInit( self ):
		'''
		Initializes our autonomous mode
		'''
		self.nt_vision.putBoolean('isTeleop', False)

		self.gyro.reset( )

		# Get the driver-selected auto mode from SmartDashboard and start it
		self.oi.auto_choose.getSelected( ).start( )
```

This seems to match our code from last year, but it's very possible this is user error.

virtuald, 2017-03-29:
Well... that should work.

virtuald, 2017-03-29:
I'm curious, does the selection work in the pyfrc simulator?

virtuald, 2017-03-29:
I just added some tests in 2fb1e46dca8ff0c9d12ed9dddaf23331dda454ce, and SendableChooser works fine. There must be something else going on in your robot code. I'm happy to try and figure it out with you, but I don't think the issue is with SendableChooser.

closed 2017-03-29 by virtuald

vScourge, 2017-03-29:
It does seem to work fine in the simulator.  I can't say why our robot code would behave differently.  I'll poke at it some more tonight.  Since it seemed to work fine on the field at competition I'm not terribly worried about it.  Thanks for checking it out.