robotpy/robotpy-wpilib #265: How to run a CommandGroup conditionally from within another CommandGroup

czhao39, 2017-04-07:
How can I conditionally run a `CommandGroup` from within another `CommandGroup` based on if a switch is pressed? It seems like all the `addSequential()`s and `addParallel()`s have to be within the constructor rather than in `initialize()`, since otherwise both the parent and the child `CommandGroup`s `require` the same subsystems. The problem is, I want the switch to be checked when the child `CommandGroup` is started, not when it is scheduled. I want something like the following:

>self.addSequential(moveCommand())
if the switch is pressed:
    self.addSequential(childCommandGroup())

However, the switch is being checked at the very beginning when everything is being scheduled, not after `moveCommand()` has been run.

Twinters007, 2017-04-07:
It might be possible to add the commandGroup anyway, and for each command, if the switch isn't pressed, have isFinished return true? A bit messy though...

czhao39, 2017-04-07:
Hm, I was hoping I wouldn't have to create a variable for every command run in the child CommandGroup that tells it whether to check for the switch. Thanks though; I'll implement that for now. If there is a more elegant solution, please let me know!

virtuald, 2017-04-07:
I would ask this question on ChiefDelphi, as it's not specific to RobotPy, and there are lots of Command programming experts there.

closed 2017-04-07 by virtuald