robotpy/robotpy-wpilib #270: jupyter can't import hal_impl.functions

ariovistus, 2017-06-03:
I'm trying to write a simulation in jupyter that makes use of robotpy, and I need to override how Timer knows what time it is. Looks like that's hal_impl.functions.hooks.getFPGATime. But jupyter throws an exception when I try to import it. 

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/ellery/.virtualenvs/plots/lib/python3.4/site-packages/hal_impl/functions.py", line 2, in <module>
    from hal import constants
  File "/home/ellery/.virtualenvs/plots/lib/python3.4/site-packages/hal/__init__.py", line 1, in <module>
    from .functions import *
  File "/home/ellery/.virtualenvs/plots/lib/python3.4/site-packages/hal/functions.py", line 9, in <module>
    from hal_impl.fndef import _RETFUNC, _THUNKFUNC, _VAR, _dll, sleep
  File "/home/ellery/.virtualenvs/plots/lib/python3.4/site-packages/hal_impl/fndef.py", line 11, in <module>
    from . import functions as _dll
```

what should I be doing here?

virtuald, 2017-06-04:
As you can see, there's a built in hooking mechanism you can use... it's not 100% clear what you did to import, but I think the order of imports is something like this (and we should try to address some of these circular dependencies):

```
import hal
import hal_impl
```

* https://github.com/robotpy/robotpy-wpilib/blob/master/hal-sim/hal_impl/sim_hooks.py#L7 

You should look at how pyfrc does it (and I suspect you will want to use some of that code). In particular:

* https://github.com/robotpy/pyfrc/blob/master/lib/pyfrc/test_support/pytest_plugin.py#L36

You will almost certainly not want to write your own implementation of this stuff, but rather reuse the pyfrc pieces. Obviously the documentation is lacking, but that's something we can work towards addressing.

Here's other things from pyfrc...

* https://github.com/robotpy/pyfrc/blob/master/lib/pyfrc/test_support/controller.py
* https://github.com/robotpy/pyfrc/blob/master/lib/pyfrc/test_support/fake_time.py

closed 2017-06-04 by virtuald

ariovistus, 2017-06-04:
```
from hal_impl import functions
```

is what I did. pretty much any way I try to access hal_impl.functions it throws exceptions

..and 

```
import hal
import hal_impl
from hal_impl import functions
```

actually does work. Is it just me, or is this unintuitive?

virtuald, 2017-06-04:
There are circular dependencies that occur when you don't import `hal` first. It is a bit unintuitive, and the circular dependency should probably be fixed.