robotpy/robotpy-wpilib #275: Hal data not entered properly or possibly at all on the command framework

amboyscout, 2017-08-15:


virtuald, 2017-08-15:
You'll need to clarify a bit here. The command framework doesn't cross the HAL boundary, so it wouldn't be in the HAL dictionary.

amboyscout, 2017-08-15:
Data such as PWM motors and CANTalons are not edited into the hal dictionary when using command code. Hal also does not update the user_program_state when simulating a command based program. 
@virtuald

amboyscout, 2017-08-16:
Reason for this problem has been discovered: Any hal_data initiated before robotInit is reset to default.

virtuald, 2017-08-16:
Oh, this is because the command examples use globals? Sigh.

virtuald, 2017-08-16:
We really should have some mechanism that allows the command based code to work without needing globals. This isn't Java after all.

Twinters007, 2017-08-18:
#195 should fix all our problems with command based

virtuald, 2017-09-17:
Closing this as it's technically not our problem -- but we do need to fix the command-based examples to not use globals (@FRC2539 do you have any ideas for this?) . I've opened up a separate issue over in the examples repo.

closed 2017-09-17 by virtuald