robotpy/robotpy-wpilib #276: Update code for 2018

virtuald, 2017-09-17:
This is a broad issue, to just be a pointer to something that needs to get done.

For those of you new to this project, every year we have to update RobotPy's source code to match whatever changes they made to WPILib. It's a bit time consuming and a very manual process, and last year I think it took us about a month to do it.

We have some tools that help out with this, if you're interested in helping out then check out our porting guide at http://robotpy.readthedocs.io/en/stable/dev/porting.html . If the instructions don't make sense, please comment here so we can update them.

virtuald, 2017-09-18:
Similar to last year, if you want to work on this, the way it works is something like so:

* Comment on this issue saying which file(s) you are working on
* Do the file(s) using the instructions in the porting guide
* Make a PR

Try not to say you're doing a file if you're not going to do it *now*, that way someone else can do it if they have free time.

virtuald, 2017-09-19:
Looks like the wpilib changes broke git source-track, I'll take a look.

virtuald, 2017-09-19:
Ok, if you upgrade to git-source-track 0.3.2 or later, then it should all work properly now.

ArthurAllshire, 2017-09-19:
Currently working on updating `adxl345_i2c.py`

Twinters007, 2017-09-21:
`robotdrive.py`

ArthurAllshire, 2017-09-24:
Working on updating all the PEED-related files/objects

ArthurAllshire, 2017-09-27:
`interfaces/*.py` ✓

ArthurAllshire, 2017-09-27:
`digitalsource.py` ✓
`digitalglitchfilter.py`✓
`controllerpower.py` ✓
`driverstation.py` WIP
Will submit a PR once the latter is done.

ariovistus, 2017-10-03:
guess I'll try my hand at 
solenoid.py
solenoidbase.py

virtuald, 2017-10-03:
Just FYI, I'm working on finishing up the ntcore 2018 updates this week. I have all the changes made now, but I need to fix/update the tests. Not going to merge any PRs until pynetworktables is updated -- but please continue to update files despite that!

ariovistus, 2017-10-04:
are they publicly available so that we could look through them to anticipate the what the new calls will look like?

ariovistus, 2017-10-04:
grabbing command/

virtuald, 2017-10-04:
I've been pushing them to https://github.com/virtuald/pynetworktables in the 'tmp' branch... but expect it all to be rebased/squashed soon. Also, I actually forgot to implement the high-level entry stuff, which is what WPILib is using now in the various table listener things...

ariovistus, 2017-10-07:
grabbing 
livewindowsendable.py
pidcontroller.py
relay.py
sendablechooser.py

virtuald, 2017-10-09:
First stage of ntcore changes is done, just have to do the user facing stuff next and then we can unstick the PR jam. Sorry for the delays, didn't realize it was going to take so long.

ArthurAllshire, 2017-10-18:
buttons/*.py

virtuald, 2017-10-18:
Almost done with the user facing stuff. Any day now!

virtuald, 2017-10-23:
ntcore changes merged, I'm fixing the tests and then we'll open the PR queue again.

Twinters007, 2017-10-23:
😩

virtuald, 2017-10-23:
Alright, everything is updated and should be working... folks with open pull requests, please rebase and fix your PRs (if necessary), and I'll merge them once the tests pass.

Twinters007, 2017-10-23:
robotdrive and related changes

ariovistus, 2017-10-24:
pynetworktables doesn't seem to have anything equivalent to https://github.com/wpilibsuite/ntcore/blob/5ab20bb27c9720304f3ca2a6d4eecd918e27555e/src/main/native/include/networktables/EntryListenerFlags.h
?

could just use NT_NOTIFY_IMMEDIATE, but that would look a bit weird when comparing with upstream code in future

virtuald, 2017-10-24:
[It does](https://github.com/robotpy/pynetworktables/blob/master/networktables/instance.py#L102)

ariovistus, 2017-10-25:
digitalinput.py
digitaloutput.py

ariovistus, 2017-10-29:
@ArthurAllshire my PR is failing because pid controller needs to be updated. do you have a lock on that, or shall I do it?

ArthurAllshire, 2017-10-30:
@arilotter I don't have a lock on it - I think i avoided doing that when I worked on the PID files cause NT wasn't ready yet and it depended on it.

ariovistus, 2017-10-31:
pidcontroller.py

ariovistus, 2017-11-22:
builtinaccelerometer.py
compressor.py
doublesolenoid.py
encoder.py
filter.py
geartooth.py
gyrobase.py
lineardigitalfilter.py
powerdistributionpanel.py
resource.py
sensorbase.py

virtuald, 2017-11-30:
FYI, I'm expecting changes to Sendable/SensorBase coming from wpilib in the near future, so there will be some changes to the livewindow stuff.

virtuald, 2017-11-30:
Also, I haven't had time to run anything that needs diagnostics, but this weekend I'll have time.

virtuald, 2017-12-05:
Looks like those changes I was expecting have landed, so we'll need to change everything around yet again... but hopefully it'll be a lot less work going forward?

https://github.com/wpilibsuite/allwpilib/commit/f9bece2ffbf7a77ccd11038b24d3d82a9118de0a

ariovistus, 2017-12-05:
I was really wondering why a bunch of components didn't have listeners set up

virtuald, 2017-12-05:
Done.

ariovistus, 2017-12-06:
SendableBuilderImpl.java

ariovistus, 2017-12-06:
NetworkTableEntry.isValid does not exist in pynetworktables, how to handle?

virtuald, 2017-12-06:
Add it?

virtuald, 2017-12-06:
Actually, I don't think isValid has an equivalent in pynetworktables... I believe entries (if they exist) will continue to always be usable.

auscompgeek, 2017-12-06:
Dibs RobotBase.

virtuald, 2017-12-06:
Existing PRs will need to be rebased to include the sphinx documentation tests, so that we don't keep breaking the doc formatting.

ariovistus, 2017-12-07:
SendableBase.java

ariovistus, 2017-12-09:
livewindow.py

Twinters007, 2017-12-12:
In order to prevent merge conflicts and whatnot, what if we said we'd update the tracking commit every time they release a new beta?

virtuald, 2017-12-23:
I'm taking the SPI changes for the newest HAL. Started on it, but need to sleep now.

virtuald, 2017-12-24:
SPI needs Notifier, and we've been talking about it, so I'm going to grab that too.

auscompgeek, 2017-12-26:
Taking the things that the examples use that still need the LiveWindow changes (that are raising DeprecationWarnings), namely:

- Encoder
- AnalogGyro
- DoubleSolenoid
- ADXRS450_Gyro

ariovistus, 2017-12-30:
dibs on notifier based pid controller

twinters007, 2017-12-30:
@ariovistus wait until dustin determines why wpilibj doesn't use it

ariovistus, 2017-12-30:
too late :)

ariovistus, 2017-12-30:
interestingly, by my tests (https://github.com/ariovistus/hal-notifier-tests/blob/master/test10.py), notifier based pid controller doesn't seem to be any better than the existing implementation for period of 10 ms or greater

virtuald, 2017-12-31:
@ariovistus did you test Java's notifier performance in addition to C++'s? That would also be interesting.

Also, how quickly does your executing function execute? That could potentially be a limiting factor.

ariovistus, 2017-12-31:
Haven't tried java. I did try spinning the executing function for 20 microseconds in python before returning. It seemed to make less of a difference than file IO.

virtuald, 2017-12-31:
I'm loathe to include numpy as a dependency unless we have to just because it means more things that could break (though, I guess cscore requires numpy).... if you went that route, I'd almost want it to be conditional, but I'm not sure that makes sense either.

ariovistus, 2018-01-01:
did some timing tests for LinearDigitalFilter: https://github.com/ariovistus/hal-notifier-tests/tree/master/filter
numpy doesn't add value unless you have a lot of gains. Not too familiar with the theory behind these things, but I suspect you aren't going to have a lot of gains.

twinters007, 2018-01-01:
💪

virtuald, 2018-01-06:
I think we're done here, calling it good for now.

closed 2018-01-06 by virtuald