robotpy/robotpy-wpilib #291: gyro not updating in sim

ariovistus, 2017-10-22:
take the physics example, add this line of code to an appropriate spot in teleop periodic:

```
     print(self.gyro.getAngle())
```

output is always 0.0

Either I'm doing something wrong, or have old versions of something, or hal sim is broken.

pip freeze:
```
py==1.4.34
pyfrc==2017.1.5
pygame==1.9.3
pynetconsole==1.3.1
pynetworktables==2017.0.8
pytest==3.2.3
robotpy-hal-base==2017.1.2
robotpy-hal-sim==2017.1.2
robotpy-installer==2017.1.4
robotpy-wpilib-utilities==2017.0.11
wpilib==2017.1.2
```

I notice that physics/core.py:PhysicsEngine._update_gyros is updating `hal_data['analog_in'][gyro_ch]['accumulator_value']`, but hal_imple/functions.py:getAnalogGyroAngle is reading from `hal_data['analog_gyro'][handle.pin]['angle']`; is there some processing that happens/should happen elsewhere, or are these out of sync?

virtuald, 2017-10-22:
They sound out of sync.

virtuald, 2017-10-22:
I bet this happened last year when they made those HAL updates to improve the APIs. Good catch.

closed 2018-02-16 by virtuald