robotpy/robotpy-wpilib #348: Are the mecanum wheel functions actually for mecanum drive trains?

RustyRaptor, 2017-12-05:
I know this question seems strange but the documentation describes a mecanum drive train as such:
```
A method for driving with Mecanum wheeled robots. There are 4 wheels
        on the robot, arranged so that the front and back wheels are toed in
        45 degrees.  When looking at the wheels from the top, the roller
        axles should form an X across the robot.
```

This doesn't seem to be the case with mecanum wheels. As seen in any demonstration or diagram they are not toed in at 45 degree angles. They re placed like regular wheels. The ones which ARE toed in at 45 degree angles are the omni wheels. 
Mecanum: https://www.youtube.com/watch?v=Q4esxUwpzWQ
Omni: https://www.youtube.com/watch?v=5vJCucpVdX0

I was just about to test the code on our mecanum drive train then I noticed what it says in the docs. So I am not sure if this will actually work for mecanum or not. I may need to just convert the mecanum code on the FRC site.

virtuald, 2017-12-05:
I'm honestly not sure, we copy our documentation from https://github.com/wpilibsuite/allwpilib... I would post your question there.

Twinters007, 2017-12-06:
It's the rollers that they are talking about. The rollers on the mecanum wheels should go inwards for proper functionality.

Twinters007, 2017-12-06:
```
    Mecanum drives are rectangular with one wheel on each corner. Each wheel has rollers toed in
    45 degrees toward the front or back. When looking at the wheels from the top, the roller axles
    should form an X across the robot.
``` 
This is the documentation in drive/mecanumdrive.py

Which means we have a docs issue

auscompgeek, 2017-12-06:
We don't have a docs issue here. @RustyRaptor is reading the 2017 (stable) documentation. Right?

Twinters007, 2017-12-06:
Latest doesn't have it either, and latest (theoretically?) should have the current docs, regardless if a tag has been pushed, right?

auscompgeek, 2017-12-06:
Known build issues on RTD regarding dependencies which nobody has been bothered to fix yet.

RustyRaptor, 2017-12-06:
No, I was looking latest. Though both latest and stable seem to have the same comments on that. I will have a look at the docs we copy from to see if I can open the issue there.

RustyRaptor, 2017-12-06:
It seems we don't have the same comments as they do. 
Theirs (search the page for '45' and read that comment): https://github.com/wpilibsuite/allwpilib/blob/f9bece2ffbf7a77ccd11038b24d3d82a9118de0a/wpilibc/src/main/native/include/Drive/MecanumDrive.h

RobotPy Docs (search 'mecanum' or '45'): http://robotpy.readthedocs.io/projects/wpilib/en/stable/_modules/wpilib/robotdrive.html 

If this is just a documentation error and it's not actually coded for Omni wheels that's a relief to me. Means I can take a break until next meeting :') 

Thank you all for your quick responses. By the way, I was wondering what is up with the RobotPY Gazebo plugins? I understand it's not being maintained so much but I would love to be able to use it. Are there any plans from anyone to continue working on it? I am not exactly sure how to even use it. I think I might make it a project for my team to study how it works and write some documentation for it.

auscompgeek, 2017-12-06:
You're reading the 2018 WPILib (currently in beta), with the new RobotDriveBase changes. Those changes have been reflected in git, but nobody has gotten around to fixing the RTD build dependency issue.

virtuald, 2017-12-06:
The Gazebo plugins are looking for someone to give them some love. When I messed with Gazebo in 2015, using their simulation stuff wasn't particularly useful. I've found that RobotPy's low-fidelity simulator works well enough for just about anything I need.

virtuald, 2017-12-06:
Also, keep in mind that we copy the Java code -- which while it should have the same comments as the C++ code, doesn't always match exactly.

You can see we match the comment you cited anyways... mmm.. no we don't. Maybe the docs got updated recently? In either case, 2018 is still a work in progress.

closed 2017-12-06 by virtuald