robotpy/robotpy-wpilib #355: Phase out 3.4

Twinters007, 2017-12-08:
Since we all seem to agree that there's no real need to keep supporting it, we should phase it out.

virtuald, 2017-12-11:
@robotpy/developers (and anyone else) let's vote on this -- get rid of Python 3.4 support now or keep it?

I vote indifferent, preferring to get rid of it if it makes life easier.

ariovistus, 2017-12-11:
I vote drop it

Twinters007, 2017-12-11:
I wouldn't even be opposed to only supporting 3.6...

auscompgeek, 2017-12-12:
I'm indifferent about this.

Let's not be too hasty about dropping 3.5 support though. PyPy3 is still on 3.5 (and even then it's a beta for x86-64 only 😢).

virtuald, 2017-12-12:
I am opposed to dropping 3.5.

Twinters007, 2017-12-12:
I'm indifferent to 3.5, but i don't see how we can solve #157 with 3.4 support

auscompgeek, 2017-12-12:
Backports are a thing.

Twinters007, 2017-12-12:
I don't know what those are

virtuald, 2017-12-22:
I think the results are in, sounds like it's out.

auscompgeek, 2017-12-31:
Given we've managed to sneak in code that only works on Python 3.5+ now (writing docstrings of properties), I think this now means that we no longer support Python 3.4 at all.

closed 2018-01-20 by virtuald