robotpy/robotpy-wpilib #377: Add helper for HAL finalizers

virtuald, 2017-12-23:
Maybe some kind of magic property, or support directly in hal... let's make it easier to get it right.

ariovistus, 2017-12-23:
maybe something like

`handle, finalizer = hal.initializePWMPort(port, obj)`

with the semantics that finalizer will be called when obj is gc'd (pass obj=None to skip this behavior).
or finalizer could be a method of handle

virtuald, 2017-12-24:
I've changed my mind. I propose the following:

* Get rid of all current weakref finalizer things for handles. If you call free() and try to use the object after that... you get what you deserve
  * If we really cared, we could rename all free functions to `_free`
* Anything that needs cleanup needs a `free()` function (even if not defined in wpilib)
* The instance gets added to `Resource._add_global_resource` or `Resource._allocate`
* `Resource` will add a weakref.finalize that calls the free function and use that instead of calling free directly as it currently does

This will make future cleanup a lot less error prone in the future, and remove a lot of copy/paste related to handles.

auscompgeek, 2017-12-26:
I wonder whether we should use `__del__`.

virtuald, 2018-12-07:
If we added `__del__`, then it would need to call `close()`, which would then need to be safe to call multiple times.

virtuald, 2020-01-19:
OBE.

closed 2020-01-19 by virtuald