robotpy/robotpy-wpilib #418: Exception in thread FRCDriverStation

VergilTheHuragok, 2018-01-10:
Whenever I reboot the robot, the DriverStation prints off [this stacktrace.](https://github.com/robotpy/robotpy-wpilib/files/1617210/crash.txt)  
  
The robot seems to continue running despite this crash although it does not seem to register any input from the gamepad afterward. However, considering that the gamepad works fine when running on the sim, I have a feeling robotpy may not be at fault for that problem. Then again, the sim has yet to experience this crash as far as I can tell.  
  
Anyways, [here's my current code](https://github.com/VergilTheHuragok/WIP_2017_Python_FRC_Robot_Team_3172/blob/master/robot.py) if it is needed.

auscompgeek, 2018-01-10:
Hi @VergilTheHuragok, thanks for reporting this. I guess we need to do more testing on roboRIOs.

This issue should be fixed by #419 once it's merged.

(If your team would like to test my fix, please be my guest; I would appreciate it as I don't have a roboRIO with me right now. You can use the `devtools/build_and_deploy.sh` bash script, whilst in a git bash window, from the git repo to deploy the fix to your robot.)

VergilTheHuragok, 2018-01-10:
I would love to try it out but I can't seem to get build_and_deploy.sh to work (I am pretty new to git/gnu/msys). When I run it with MSYS, it prints:  
  
.\build_and_deploy.sh: line 11: dirname: command not found
.\build_and_deploy.sh: line 11: cd: HOME not set  
  
I am not in a hurry, however, so I have no problem waiting for the merge if necessary.

auscompgeek, 2018-01-10:
Huh, MSYS doesn't have dirname? That's odd.

I think you should be able to replace line 11 with `cd ${0%/*}` or similar. I also only just noticed the `devtools/_windows_env.sh` is quite outdated - you'll probably want to replace the entire body of the if statement in there with `alias python3='py -3'`.

VergilTheHuragok, 2018-01-10:
I just went ahead and entered the entire path to devtools\ which seems to have fixed that problem. 

Now it's printing:

    fatal: No names found, cannot describe anything.  
  
Thanks for the help!

auscompgeek, 2018-01-10:
Hm. How did you clone the git repo?

PR #419 (2018-01-10): Fix getMatchInfo erroring on roboRIO
_This PR has been brought to you by the nursery rhyme "I'm a Little Teapot". The patch may be short and stout._

Fixes: #418
commit f38cccdd (2018-01-10): Fix getMatchInfo erroring on roboRIO

Fixes: #418
closed 2018-01-10 by virtuald