robotpy/robotpy-wpilib #423: Problem after deploying code and robot tries to run code.

stephonlawrence, 2018-01-11:
I installed the latest image and robotpy on the robot.
I am using pycharm as my ide. The arg i'm passing when running my file named `robot.py` is `deploy`.

The console says tests are passing and deployment was successful. But when I open the driver station application, I have communication to the robot but robot code is red. Then the error in the driver station logs is:
```
ERROR  1  ERROR Unhandled <class 'TypeError'>  /usr/local/lib/python3.6/site-packages/wpilib/_impl/main.py.16:ds_except_hook 
ERROR Unhandled {}".format(exctype), True) 
ERROR Unhandled <class 'TypeError'>" 
ERROR Unhandled <class 'TypeError'>
```
This is the code in robot.py:
```
"""
    This is a good foundation to build your robot code on
"""

import wpilib
import wpilib.drive

class MyRobot(wpilib.IterativeRobot):

    def robotInit(self):
        """
        This function is called upon program startup and
        should be used for any initialization code.
        """
        self.left_motor = wpilib.Spark(0)
        self.right_motor = wpilib.Spark(1)
        self.drive = wpilib.drive.DifferentialDrive(self.left_motor, self.right_motor)
        self.stick = wpilib.Joystick(1)

    def autonomousInit(self):
        """This function is run once each time the robot enters autonomous mode."""
        self.auto_loop_counter = 0

    def autonomousPeriodic(self):
        """This function is called periodically during autonomous."""

        # Check if we've completed 100 loops (approximately 2 seconds)
        if self.auto_loop_counter < 100:
            self.drive.arcadeDrive(-0.5, 0)  # Drive forwards at half speed
            self.auto_loop_counter += 1
        else:
            self.drive.arcadeDrive(0, 0)  # Stop robot

    def teleopPeriodic(self):
        """This function is called periodically during operator control."""
        self.drive.arcadeDrive(self.stick.getY(), self.stick.getX())

    def testPeriodic(self):
        """This function is called periodically during test mode."""
        pass

if __name__ == "__main__":
    wpilib.run(MyRobot)
```
I have no other files in my directory except for .idea.
I am using a computer with Windows 7.

Can someone explain what this error means or how to solve it?

virtuald, 2018-01-11:
Which version of RobotPy do you have installed? 2018.0.4?

virtuald, 2018-01-12:
Ping?

stephonlawrence, 2018-01-12:
Sorry didn't see this until now. I will check in a few hours when I'm there, but I looked up the previous releases and based on the log file from this other issue, I think this is related to #415. Which was patched.

I'll update this in a few but also try a later release today.

Thanks.

twinters007, 2018-01-15:
@stephonlawrence any update?

stephonlawrence, 2018-01-17:
Yes that was the problem. Thanks for the help! Just needed to get latest update.

closed 2018-01-17 by stephonlawrence