robotpy/robotpy-wpilib #428: wpi_talonsrx modules' setInverted method not implemented

amorygalili, 2018-01-14:
I get this error when I try to call ctre.wpi_talonsrx.WPI_TalonSRX's setInverted method:

self = <ctre.wpi_talonsrx.WPI_TalonSRX object at 0x06626230>, invert = True

    def setInverted(self, invert: bool) -> None:
        assert isinstance(invert, bool)
>       raise NotImplementedError
E       NotImplementedError

..\..\AppData\Local\Programs\Python\Python36-32\lib\site-packages\ctre\_impl\autogen\motcontroller_sim.py:56: NotImplementedError

auscompgeek, 2018-01-15:
Yes, simulation is not implemented.

auscompgeek, 2018-01-15:
Also, issues with robotpy-ctre should be reported there, not here.

closed 2018-01-15 by auscompgeek

amorygalili, 2018-01-15:
I'm confused, is this going to be fixed?

auscompgeek, 2018-01-15:
It is intended to be fixed soon, yes. Feel free to help out, I guess?

amorygalili, 2018-01-15:
Sure, if I get to it first although I think @virtuald said he was going to look at it.