robotpy/robotpy-wpilib #436: (Question) Works with cRIO?

alicerunsonfedora, 2018-01-27:
I have a particular question about RobotPy in relation to the cRIO.

We have a bunch of old cRIOs that we use for practice bots, even though they are outdated and no longer allowed in competition. I have been looking at this and wanted to program our robot, both the main and the practice, with this port. However, I don’t know if this still supports the cRIO or if it’s restricted to the roboRIO.

Does this version of RobotPy still support the cRIO, or am I out of luck?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

twinters007, 2018-01-27:
https://github.com/robotpy/robotpy-crio

Try this

virtuald, 2018-01-27:
This particular version of RobotPy does not run on a cRIO. You can use old releases of RobotPy, found at https://github.com/robotpy/robotpy-crio/releases ... they should still work.

There are a number of differences. Most prominently, the functions are CapsCase instead of camelCase.

virtuald, 2018-01-27:
Also, if you want this particular library to work on a cRIO, it's potentially possible (but probably not worth it)... see this bug: https://github.com/robotpy/robotpy-crio/issues/8

closed 2018-01-27 by virtuald