robotpy/robotpy-wpilib #438: I2C NotImplementedError

codetheweb, 2018-01-27:
I've been trying to get communication working over I2C between our roboRIO and an Arduino, and I've run into some issues.  It appears the Arduino (sometimes?) successfully receives data, but regardless of whether or not the data goes through an error is thrown on the roboRIO side.  

Here's the code:
```python
self.arduinoC = wpilib.I2C(wpilib.I2C.Port.kOnboard, 4)
self.arduinoC.transaction(b'sbt255', 1)
```
(docs for I2C are [here](http://robotpy.readthedocs.io/projects/wpilib/en/stable/wpilib/I2C.html?highlight=i2c#wpilib.i2c.I2C.transaction))

And here's the result of local tests:
```python
self = <hal_impl.i2c_helpers.I2CSimBase object at 0x10dc90d68>, port = 0, deviceAddress = 4, dataToSend = <hal.functions.c_ubyte_Array_6 object at 0x10dc6f488>, sendSize = 6
dataReceived = <hal.functions.c_ubyte_Array_1 object at 0x10dc6fbf8>, receiveSize = 1

    def transactionI2C(self, port, deviceAddress, dataToSend, sendSize, dataReceived, receiveSize):
        '''
                To give data back use ``data_received``::
    
                    data_received[:] = [1,2,3...]
    
                :returns: number of bytes returned
            '''
>       raise NotImplementedError
E       NotImplementedError
```

I'm guessing this is the result of 2018 changes to WPILib, but I'm not familiar enough with it to confirm.

virtuald, 2018-01-27:
What error shows up on the RoboRIO side? It'll be a different error I imagine. I don't know if we actually updated the I2C stuff for 2018, so that might be the issue. Seeing the error would be useful however.

When using I2C in simulation/testing, you have to implement the device side of the simulation (which is why you're getting a NotImplementedError). We don't really have a document yet describing how you would go about this... but I could put some documentation together tonight.

codetheweb, 2018-01-27:
Ok, I didn't realize the error was a result of it being in a simulation.

Unfortunately I won't have access to the robot until Monday, so I'll update you then.

virtuald, 2018-01-29:
I added a bunch of documentation, and the error messages will be more helpful in future releases. Hope that helps!

virtuald, 2018-01-29:
You can find it at http://robotpy.readthedocs.io/projects/wpilib/en/latest/hal.html#i2c-spi-simulation-helpers

codetheweb, 2018-01-29:
Here's the error on the robot side.  It appears something throws an error on success(?)

``` python
Traceback (most recent call last):
   File "/usr/local/lib/python3.6/site-packages/wpilib/robotbase.py", line 168, in main
     robot.startCompetition()
   File "/usr/local/lib/python3.6/site-packages/wpilib/samplerobot.py", line 164, in startCompetition
     self.operatorControl()
   File "/home/lvuser/py/robot.py", line 57, in operatorControl
     self.lights.stagger('blue', True, 255)
   File "/home/lvuser/py/components/lights.py", line 15, in stagger
     self.arduinoC.transaction(b'sbt255\n', 0)
   File "/usr/local/lib/python3.6/site-packages/wpilib/i2c.py", line 95, in transaction
     dataToSend, receiveSize)
   File "/usr/local/lib/python3.6/site-packages/hal/functions.py", line 443, in transactionI2C
     raise IOError(_os.strerror(C.get_errno()))
 OSError: Success
ERROR  1  Unhandled exception  /usr/local/lib/python3.6/site-packages/wpilib/_impl/main.py.136:run
 
 
 Locals at innermost frame:
 
 { 'dataToSend': b'sbt255\n',
   'deviceAddress': 4,
   'port': 0,
   'receiveSize': 0,
   'recv_buffer': <hal.functions.c_ubyte_Array_0 object at 0xb473df30>,
   'rv': -1,
   'sendSize': 7,
   'send_buffer': <hal.functions.c_ubyte_Array_7 object at 0xb473dee0>}
 
 
 Error at /usr/local/lib/python3.6/site-packages/wpilib/_impl/main.py.136:run: Unhandled exception
 :
 Traceback (most recent call last):
   File "/home/lvuser/py/robot.py", line 69, in <module>
     wpilib.run(Randy)
   File "/usr/local/lib/python3.6/site-packages/wpilib/_impl/main.py", line 136, in run
     retval = options.cmdobj.run(options, robot_class, **kwargs)
   File "/usr/local/lib/python3.6/site-packages/wpilib/robotbase.py", line 168, in main
     robot.startCompetition()
   File "/usr/local/lib/python3.6/site-packages/wpilib/samplerobot.py", line 164, in startCompetition
     self.operatorControl()
   File "/home/lvuser/py/robot.py", line 57, in operatorControl
     self.lights.stagger('blue', True, 255)
   File "/home/lvuser/py/components/lights.py", line 15, in stagger
     self.arduinoC.transaction(b'sbt255\n', 0)
   File "/usr/local/lib/python3.6/site-packages/wpilib/i2c.py", line 95, in transaction
     dataToSend, receiveSize)
Warning  1  Robots should not quit, but yours did!  
   File "/usr/local/lib/python3.6/site-packages/hal/functions.py", line 443, in transactionI2C
     raise IOError(_os.strerror(C.get_errno()))
 OSError: Success
 
 15:29:06:613 WARNING : wpilib.ds           : Robots should not quit, but yours did!
 Robots should not quit, but yours did!
```

virtuald, 2018-01-30:
Hm, sounds like a bug. I won't be able to look tonight, but perhaps tomorrow. I don't have i2c set up at the moment... but I might be able to find time tomorrow with a device I have.

codetheweb, 2018-01-31:
I believe I might have found the issue.  Every other function in the I2C class wraps the actual HAL function in a try/except block, like so:
```python
def write(self, registerAddress, data):
        """Execute a write transaction with the device.

        Write a single byte to a register on a device and wait until the
        transaction is complete.

        :param registerAddress:
            The address of the register on the device to be written.
        :param data: The byte to write to the register on the device.
        :returns: Transfer Aborted... False for success, True for aborted.
        """
        try:
            hal.writeI2C(self.port, self.deviceAddress, [registerAddress, data])
        except IOError:
            return True
        return False
```

This has the effect that if/when the HAL function throws an IOError, `write()` returns true.  I think this is needed for `transaction()` as well.  Current `transaction()` function:
```python
def transaction(self, dataToSend, receiveSize):
        """Generic transaction.

        This is a lower-level interface to the I2C hardware giving you more
        control over each transaction.

        :param dataToSend: Buffer of data to send as part of the transaction.
        :type dataToSend: iterable of bytes
        :param receiveSize: Number of bytes to read from the device.
        :type receiveSize: int
        :returns: Data received from the device.
        :rtype: iterable of bytes
        """
        return hal.transactionI2C(self.port, self.deviceAddress,
                                  dataToSend, receiveSize)
```

auscompgeek, 2018-01-31:
How would you propose writing `I2C.transaction()` such that we:
* do not break user code, and
* let the user know that the transaction failed?

I don't know why the error says "success" though. That seems to imply that the real errno set in `HAL_TransactionI2C()` is being reset somewhere, which isn't good.

codetheweb, 2018-01-31:
From [your team's code](https://github.com/thedropbears/pypowerup/blob/5ceb026af1807b917e316f9675b03d5a597fafbf/utilities/bno055.py#L244) it appears you're using `write()`, not `transaction()`, which has also been working for us without problems.

I would vote to just throw in a block similar to the one in `write()`:
```python
def transaction(self, dataToSend, receiveSize):
        """Generic transaction.

        This is a lower-level interface to the I2C hardware giving you more
        control over each transaction.

        :param dataToSend: Buffer of data to send as part of the transaction.
        :type dataToSend: iterable of bytes
        :param receiveSize: Number of bytes to read from the device.
        :type receiveSize: int
        :returns: Data received from the device.
        :rtype: iterable of bytes
        """

        try:
            return hal.transactionI2C(self.port, self.deviceAddress,
                                            dataToSend, receiveSize)
        except IOError:
            return True
        return False
```

The return values seem backwards (returning true on "error" and false on no error), but they're consistent with the rest of the I2C class.

auscompgeek, 2018-02-01:
Huh, I thought we were using transaction as well. I guess not.

Your suggestion throws away the data received, so that's a no go.

codetheweb, 2018-02-01:
🙄 Sorry, I didn't even think about that as we're not receiving any data in our application.

I'm not really sure what to do in that case.  (Although if `hal.transactionI2C()` throws an error, we can't get a return value anyways.)

auscompgeek, 2018-02-01:
Wait. I2C.read() does call I2C.transaction(), so my team _is_ (indirectly) using I2C.transaction().

Yeah, I dunno what's going on.

codetheweb, 2018-02-01:
Here's something interesting.  If `receiveSize` for `transaction(dataToSend, receiveSize)` is a non-zero number, it doesn't throw any errors.  This is why `read()` works.  Not sure why this is, but I guess we can just do `transaction(b'data', 1)` for now.

auscompgeek, 2018-02-02:
Hm, I guess it sorta makes sense? You were saying "hey, I want to read and write, but I want to read 0 bytes", which didn't quite make much sense.

codetheweb, 2018-02-02:
I guess.  Actually, I'm not sure why we're not just using `writeBulk()`... it would make a lot more sense.  I'll open a pull request with updated docs to make it more clear.

auscompgeek, 2018-02-10:
I think we should keep this open and investigate why the errno is 0 (success) when there is an error. I'm still puzzled as to how the errno would not be indicative of the error…

(Just saw this myself by not having the intended thing on the I2C bus.)

commit 5bc4534b (2018-01-29): Document I2C/SPI simulation helpers

- Fixes #438
PR #580 (2019-02-01): hal-roborio: Fix getting errno
Fixes: #438
commit 5d968bd4 (2019-02-01): hal-roborio: Fix getting errno

Fixes: #438
closed 2019-02-01 by virtuald