robotpy/robotpy-wpilib #441: LiveWindow.updateValues taking 64 mSec in teleop

dangyogi, 2018-02-01:
This is running on the roboRio with wpilib version 2018.0.11.

iterativerobotbase.py:148(loopFunc)       ->       1    0.000    0.000  livewindow.py:83(setEnabled)
                                                                         19    0.005    1.217  livewindow.py:301(updateValues)

1.217 / 19 is 64 mSec.

The attached "profile" file (had to zip to be able to do the attachment) was generated by cProfiler.dump_stats.  You can examine it with:

$ python3 -m pstats profile

This profile was generated by enabling the profiler during an autonomous routine that was triggered by pressing a joystick button.  The autonomous routine enabled the profiler at the beginning and disabled it at the end.  So this does _not_ include extraneous, e.g. initialization, code.  Note that the robotPeriodic method was run 19 times during the 3.3 seconds that this ran.  Thus, the robotPeriodic method is being called every 174 mSec (on average, 3.3/19).  My code (modularrobot.py:284(robotPeriodic)) is taking 75 mSec (1.43/19), which seems to largely be because it's writing ~20 values to the SmartDashboard.

This leaves 99 mSec overhead taken by the wpilib.  64 mSec of this is LiveWindow.updateValues.  That leaves 35 mSec per cycle.  It looks like the remaining time being reported by the profiler is only ~3 mSec per cycle.  That leaves 32 mSec per cycle unaccounted for.  My guess is that this time is being taken in other threads.  Outside of the default threads used by the wpilib (e.g., for network tables), the only additional threads that I've caused to be created are that I have 2 PIDControllers, which were not used in this test -- though I believe that these threads are still running.  (Multiple cores are pretty much useless with Python due to the GIL lock, so the additional thread overhead would be running on the same cpu core as the main program).

So the additional 32 mSec seems like an additional performance problem.  Should this be filed as another bug?

[profile.zip](https://github.com/robotpy/robotpy-wpilib/files/1683781/profile.zip)

virtuald, 2018-02-12:
Sorry for the delay, I'd like to take a closer look into this tomorrow. We did just make some changes recently that affected the loop timing, which may positively affect this issue. Have you tried this recently after updating?

Also, what kinds of wpilib objects were you using when this occurred? I'd like to try to duplicate this -- perhaps you could just post your object initialization code here?

dangyogi, 2018-02-12:
Here are the current 3 robotInit functions.  There have probably been some
changes, but I think it's the same set of wpilib objects as when I reported
the problem.

Also using SmartDashboard and Preferences.

I've highlighted the wpilib classes.  The erl_is_dead.Encoder is derived
from the Encoder in the wpilib.

I haven't tried this with the new wpilib.  Right now, I'm going through a
big refactoring, so it will be a few days before I can get new data.

Let me know if there is anything else I can do.

-Bruce

-----------------------------------

@erl_is_dead.robotInit
def create_drivetrain():
    robot = erl_is_dead.get_robot()

    robot.left_sc = erl_is_dead.SingleController("left_sc",
                      VictorSP(ports.pwm.left_drive_motor),
                      ports.pdp.left_drive_motor)
    robot.left_vc = erl_is_dead.VoltageController("left_vc", robot.left_sc)
    robot.left_encoder = erl_is_dead.Encoder("left_encoder",
                                             ports.dio.left_encoder_A,
                                             ports.dio.left_encoder_B,
                                             reverse=True)
    left_ac = erl_is_dead.AccelerationController("left_ac", robot.left_vc,
                                                 robot.left_encoder)

    robot.right_sc = erl_is_dead.SingleController("right_sc",
                       VictorSP(ports.pwm.right_drive_motor),
                       ports.pdp.right_drive_motor)
    robot.right_vc = erl_is_dead.VoltageController("right_vc",
robot.right_sc)
    robot.right_encoder = erl_is_dead.Encoder("right_encoder",
                                              ports.dio.right_encoder_A,
                                              ports.dio.right_encoder_B,
                                              reverse=True)
    right_ac = erl_is_dead.AccelerationController("right_ac",
robot.right_vc,
                                                  robot.right_encoder)

    robot.skid_drive = erl_is_dead.SkidDrive("skid_drive", left_ac,
right_ac,
                                             ADXRS450_Gyro())
    robot.physics = robot.skid_drive.physics

-----------------------------------------------------------

@robotInit
def init_driver_controls():
    robot = get_robot()

    robot.xbox_0 = XboxController(0)

    # Must negate all Y joystick axes!
    #robot.speed_control = negate(partial(robot.xbox_0.getY, Hand.kRight))
# 5
    robot.forward_speed_control = partial(robot.xbox_0.getTriggerAxis,
                                          Hand.kRight) # 3
    robot.reverse_speed_control = partial(robot.xbox_0.getTriggerAxis,
                                          Hand.kLeft) # 2

    robot.turning_rate_control = partial(robot.xbox_0.getX, Hand.kRight) # 4
    robot.turning_radius_control = partial(robot.xbox_0.getX, Hand.kLeft) #
0
[... 72 lines cut ...]
    #                                            robot.xbox_0,
    #                                            JoystickButton.Xbox_B,
    #                                            'test')
    robot.turn_test_button = JoystickButton('turn_test_button',
robot.xbox_0,
                                            JoystickButton.Xbox_X,
                                            'test', 'teleop')

robot.reset_button.registerRunOnceOnPressed(robot.skid_drive.reset_encoders,
                                                'all')
    robot.bump_test_button.registerRunOnceOnPressed(test_bump, 'teleop',
'test')
    robot.speed_test_button.registerRunOnceOnPressed(test_speed,
                                                     'test', 'teleop')
    #robot.turn_speed_test_button.registerRunOnceOnPressed(test_turn_speed,
    #                                                      'test')
    robot.move_test_button.registerRunOnceOnPressed(test_move, 'test',
'teleop')
    #robot.move_test_button.registerRunOnceOnPressed(forward_speeds, 'test')
    #robot.move_test_button.registerRunOnceOnPressed(test_pid_move, 'test')
    #robot.pid_turn_test_button.registerRunOnceOnPressed(test_pid_turn,
'test')
    robot.turn_test_button.registerRunOnceOnPressed(test_turn, 'test',
'teleop')
    #robot.turn_test_button.registerRunOnceOnPressed(turning_speeds, 'test')

virtuald, 2018-02-14:
Sorry for the delay. I put together a sample robot program that has similar objects as your program has, and I'm not able to duplicate the problem. Here's my minimal program:

```
#!/usr/bin/env python3

import wpilib
import wpilib.buttons

from robotpy_ext.misc.looptimer import LoopTimer

class MyRobot(wpilib.IterativeRobot):
    '''Main robot class'''
    
    def robotInit(self):
        '''Robot-wide initialization code should go here'''
        
        self.motor1 = wpilib.VictorSP(0)
        self.encoder = wpilib.Encoder(0, 1)
        
        self.motor1 = wpilib.VictorSP(1)
        self.encoder = wpilib.Encoder(2, 3)
        
        self.stick = wpilib.Joystick(0)
        
        self.buttons = [
            wpilib.buttons.JoystickButton(self.stick, i) for i in range(6)
        ]
    
    def autonomousInit(self):
        '''Called only at the beginning of autonomous mode'''
    
    def autonomousPeriodic(self):
        '''Called every 20ms in autonomous mode'''
        pass

    def disabledInit(self):
        '''Called only at the beginning of disabled mode'''
    
    def disabledPeriodic(self):
        '''Called every 20ms in disabled mode'''
        pass

    def teleopInit(self):
        '''Called only at the beginning of teleoperated mode'''
        self.looptimer = LoopTimer(self.logger)

    def teleopPeriodic(self):
        '''Called every 20ms in teleoperated mode'''
        self.looptimer.measure()

if __name__ == '__main__':
    wpilib.run(MyRobot)

```

Here's some output from the loop timer on the console... as you can see the loop period is fairly stable:

```
13:47:21:639 INFO    : robot               : Loops: 50; min: 0.018; max: 0.023; period: 1.010; avg: 0.020 
13:47:22:625 INFO    : robot               : Loops: 49; min: 0.019; max: 0.023; period: 0.987; avg: 0.020 
13:47:23:634 INFO    : robot               : Loops: 50; min: 0.019; max: 0.022; period: 1.008; avg: 0.020 
13:47:24:635 INFO    : robot               : Loops: 50; min: 0.019; max: 0.021; period: 1.001; avg: 0.020 
13:47:25:638 INFO    : robot               : Loops: 50; min: 0.018; max: 0.022; period: 1.003; avg: 0.020
```

If you can help me reproduce this then we can get it fixed. 👍

dangyogi, 2018-02-15:
OK, I'll try to take another look.  Might be after bag day though before I
can get to it.

virtuald, 2018-03-08:
I'm assuming this issue hasn't occurred for you anymore. If that changes, please reopen the issue.

nickschatz, 2018-05-16:
https://github.com/FRC3184/frc2018/tree/7140e1bace2083e7b98ccbac214143d69df0630f Version of the code where this issue occurs. Will post cProfile output when I turn on that roborio again, but from what I can tell the deepest it goes is LiveWindow.updateValues and a method or two in SendableBase iirc.

virtuald, 2018-05-17:
@nickschatz, I started your code in sim and noticed you have a PDP object ... see https://github.com/wpilibsuite/allwpilib/issues/1071 . I wonder if this issue goes away if that is removed.

virtuald, 2018-12-07:
I think the PDP was to blame for this, so it shouldn't happen anymore. If it does, file an issue in 2019.

closed 2018-12-07 by virtuald