robotpy/robotpy-wpilib #443: How to use the Game Specific Message in robotpy?

RustyRaptor, 2018-02-04:
It doesn't seem to work like the C++ version or maybe I am forgetting to do something. 

test output:
```
/home/cat/PycharmProjects/untitled1/venv/bin/python /home/cat/PycharmProjects/5613-tank-drive-2018_2/robot.py test
20:22:24:784 INFO    : wpilib              : WPILib version 2018.0.11
20:22:24:785 INFO    : wpilib              : HAL base version 2018.0.11; sim platform version 2018.0.11
20:22:24:785 INFO    : wpilib              : Running with simulated HAL.
============================= test session starts ==============================
platform linux -- Python 3.5.2, pytest-3.4.0, py-1.5.2, pluggy-0.6.0
rootdir: /home/cat/PycharmProjects/5613-tank-drive-2018_2/tests, inifile:
collected 5 items

pyfrc_test.py F....                                                      [100%]

=================================== FAILURES ===================================
_______________________________ test_autonomous ________________________________

control = <pyfrc.test_support.controller.TestController object at 0x7f6c3ef83ef0>
fake_time = <pyfrc.test_support.fake_time.FakeTime object at 0x7f6c40df4e10>
robot = <__main__.MyRobot object at 0x7f6c40ed4b38>

    def test_autonomous(control, fake_time, robot):
        '''Runs autonomous mode by itself'''
    
        # run autonomous mode for 15 seconds
        control.set_autonomous(enabled=True)
>       control.run_test(lambda tm: tm < 15)

../../untitled1/venv/lib/python3.5/site-packages/pyfrc/tests/basic.py:21: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
../../untitled1/venv/lib/python3.5/site-packages/pyfrc/test_support/controller.py:148: in run_test
    self._robot.startCompetition()
../../untitled1/venv/lib/python3.5/site-packages/wpilib/samplerobot.py:150: in startCompetition
    self.autonomous()
../robot.py:43: in autonomous
    gameData = wpilib.DriverStation.getInstance().getGameSpecificMessage()
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <wpilib.driverstation.DriverStation object at 0x7f6c4298e0b8>

    def getGameSpecificMessage(self) -> str:
        """Get the game specific message.
    
            :returns: The game specific message
            """
        with self.cacheDataMutex:
>           message = self.matchInfo.gameSpecificMessage
E           AttributeError: gameSpecificMessage

../../untitled1/venv/lib/python3.5/site-packages/wpilib/driverstation.py:502: AttributeError
====================== 1 failed, 4 passed in 1.13 seconds ======================

Process finished with exit code 1
```
Our code:
```
#!/usr/bin/env python3

import wpilib
from wpilib.drive import DifferentialDrive

class MyRobot(wpilib.SampleRobot):

    def robotInit(self):
        self.frontLeft = wpilib.Spark(0)
        self.rearLeft = wpilib.Spark(1)
        self.left = wpilib.SpeedControllerGroup(self.frontLeft, self.rearLeft)

        self.frontRight = wpilib.Spark(2)
        self.rearRight = wpilib.Spark(3)
        self.right = wpilib.SpeedControllerGroup(self.frontRight, self.rearRight)

        self.drive = DifferentialDrive(self.left, self.right)
        self.stick2 = wpilib.Joystick(0)
        self.stick = wpilib.Joystick(1)

    def disabled(self):
        while self.isDisabled():
            wpilib.Timer.delay(0.01)

    def operatorControl(self):
        timer = wpilib.Timer()
        timer.start()
        while self.isOperatorControl() and self.isEnabled():
            if self.stick2.getRawButton(1) and self.stick2.getRawButton(7):
                self.drive.arcadeDrive(self.stick2.getY() * 0.6, self.stick2.getX() * 0.7, True)
            elif self.stick2.getRawButton(7) and not self.stick2.getRawButton(1):
                self.drive.arcadeDrive(self.stick2.getY() * -0.6, self.stick2.getX() * 0.7, True)
            elif self.stick2.getRawButton(1) and not self.stick2.getRawButton(7):
                self.drive.arcadeDrive(self.stick2.getY() * 1, self.stick2.getX() * 1, True)
            else:
                self.drive.arcadeDrive(self.stick2.getY() * -1, self.stick2.getX() * 1, True)
            timer.delay(0.02)

    def autonomous(self):
        timer = wpilib.Timer()
        timer.start()
        gameData = wpilib.DriverStation.getInstance().getGameSpecificMessage()

if __name__ == '__main__':
    wpilib.run(MyRobot)
```

RustyRaptor, 2018-02-04:
Oh wait I think I figured it out. Can't tell if it's working though. Is this what I needed to do?

```
    def autonomous(self):
        timer = wpilib.Timer()
        timer.start()
        gameData = str(wpilib.DriverStation.getInstance().getGameSpecificMessage)

        if gameData[0] == 'L':
            # Do autonomous stuff
```

I see now the C++ code does this by initializing gameData as a string.

RustyRaptor, 2018-02-04:
ok the above code doesn't return any errors. Here is what it looks like now. 

```
    def autonomous(self):
        timer = wpilib.Timer()
        timer.start()
        gameData = str(wpilib.DriverStation.getInstance().getGameSpecificMessage)
        if gameData[0] == 'L':
            while self.isAutonomous() and self.isEnabled():
                self.drive.tankDrive(1,1)
        elif gameData[0] == 'R':
            while self.isAutonomous() and self.isEnabled():
                self.drive.tankDrive(-1,-1)
        else:
            while self.isAutonomous() and self.isEnabled():
                self.drive.tankDrive(0.5, -0.5)
```

however, it only runs that last else statement when I use a simulator. I am guessing I need to connect an actual driver station for it to get the game-specific message.

auscompgeek, 2018-02-04:
`DriverStation.getInstance().getGameSpecificMessage()`, same as Java.

RustyRaptor, 2018-02-04:
That doesn't seem to be working though. Check my first comment. I am calling it in autonomous and setting it to ``` gameData ```

RustyRaptor, 2018-02-04:
It does seem to work when I leave out the last parentheses and convert it to a string. Which is what is done in Java by initializing the variable as a String.

RustyRaptor, 2018-02-04:
Oh, I think I understand the problem now. In order for it to have a value at all, it needs a driver station connected? I see in DropBears code you bypassed this by checking if it is of length 3 and just pass in default values. Ok, I will just have to wait till I have access to the driver station and robot again thanks.

auscompgeek, 2018-02-04:
I'm surprised you're getting an AttributeError in the pyfrc tests… there's a fix for that sitting in #437 I guess? It should be set before your autonomous code starts though…

twinters007, 2018-02-04:
`   gameData = str(wpilib.DriverStation.getInstance().getGameSpecificMessage)` this won't work because you aren't calling getGameSpecificMessage. You're converting the function to a string

RustyRaptor, 2018-02-05:
Thanks, I think I understand now. So I am passing the function itself as a string rather than passing what the function returns. Since I removed the parentheses at the end of the function which gives it a 'self' argument.

twinters007, 2018-02-05:
correct

virtuald, 2018-02-10:
This should be addressed now.

closed 2018-02-10 by virtuald