robotpy/robotpy-wpilib #444: :inherited-members: in docs

twinters007, 2018-02-07:
With all the subclassing wpilib has, it can be very useful to see the inherited members of a class. I think we should add this feature of showing all the functions a class has.

auscompgeek, 2018-02-07:
I dunno about this. A lot of the methods in things like SensorBase aren't really relevant to actually using things like AnalogAccelerometer, for example.

virtuald, 2018-02-07:
Well. You're probably both right. Maybe we can figure out how to filter out things we don't want to propagate up? All the doc stuff is autogen'ed anyways.

auscompgeek, 2018-02-07:
Well, luckily all the base classes are listed in the docs anyway, so it's not a major problem if we bikeshed this.

virtuald, 2018-02-07:
🚴

twinters007, 2018-02-07:
I was thinking about things like SensorBase that we'd wanna hide...Maybe we cherrypick certain things? I know things like the drive bases are widely used, and it'd be nice to show `setDeadband` in `DifferentialDrive`

virtuald, 2018-02-07:
I think try it out, build it locally, and once you're happy with something then PR it and we can take a look.

twinters007, 2018-02-08:
Ok

closed 2023-11-13 by virtuald